Xilinx/Vitis-AI #1119: Error in execution of vitis_quantize.VitisQuantizer.

chikako, 2022-12-27:
I am using docker image xilinx/vitis-ai:2.5.0.1260(CPU).
I downloaded tf2_efficientnet-b0_imagenet_224_224_0.78G_2.5 from Modelzoo.
 When running conda activate vitis-ai-tensorflow2 and training the quantize model, the following error occurred.
```
  File "train_eval_h5.py", line 225, in <module>
    main()
  File "train_eval_h5.py", line 159, in main
    freeze_bn_delay=40000)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 710, in get_qat_model
    self._create_qat_model()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 343, in _create_qat_model
    mode='QAT')
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 1140, in create_quantize_model
    quantize_strategy.get_quantize_registry(), mode)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_transforms_pipeline.py", line 214, in apply
    layer_metadata).recursive_transform()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 740, in recursive_transform
    self.layer_metadata).transform()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 698, in transform
    transformed_model = keras.Model.from_config(self._config, custom_objects)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/training.py", line 2641, in from_config
    functional.reconstruct_from_config(config, custom_objects))
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/functional.py", line 1338, in reconstruct_from_config
    if process_node(layer, node_data):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/functional.py", line 1282, in process_node
    output_tensors = layer(input_tensors, **kwargs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/utils/traceback_utils.py", line 67, in error_handler
    raise e.with_traceback(filtered_tb) from None
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/layers/vitis_conv_bn.py", line 245, in build
    self._convolution_op = self.conv_layer._convolution_op
AttributeError: 'Conv2D' object has no attribute '_convolution_op'
```
This is an error in train_eval_h5.py in the code/com directory of tf2_efficientnet-b0_imagenet_224_224_0.78G_2.5.

model = vitis_quantize.VitisQuantizer(model, '8bit_tqt').get_qat_model(
              calib_dataset=calib_dataset,
              init_quant=True, 
              include_cle=False,
              train_with_bn=True,
              freeze_bn_delay=40000)

How can I get rid of this error?

chikako, 2022-12-27:
Please help me

renjiangyong, 2023-06-20:
hi @chikako ， this issue is fixed in Vitis-AI 3.0 ， you can use the latest Vitis-AI docker 3.0 to solved this issue. Thank you.

closed 2023-06-20 by renjiangyong