Xilinx/Vitis-AI #1123: Question about fast fine-tuning quantization in pytorch (accuracy loss greater than expected)

daniperfer, 2023-01-04:
Hi:
I am trying to quantize a torchvision model (it is a slightly modified version of torchvision RetinaNet model), but when I apply direct quantization I see a large accuracy loss between the original float model and the quantized model.

For this model, I am using coco evaluation metrics. The AP of the float model is 
`Average Precision (AP) @[ IoU=0.50 | area= all | maxDets=100 ] = 0.710`
and the directly quantized model:
`Average Precision (AP) @[ IoU=0.50 | area= all | maxDets=100 ] = 0.410`

Thus, I decided to try fast-finetuning quantization, following the guidelines in (https://docs.xilinx.com/r/2.0-English/ug1414-vitis-ai/vai_q_pytorch-Fast-Finetuning) and the example in (https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Quantizer/vai_q_pytorch/example/mobilenetv2_fast_finetune.py).

The documentation says that 5120 images are enough for fast-finetuning the model. However, if I configure a dataset load of 5120 imagenes for my model, I've got the following message:

`[VAIQ_NOTE]: Memory cost by fast finetuning is 117.43 G.`

`[VAIQ_WARN]: There is not enought memory for fast finetuning and this process will be ignored!.Try to use a smaller calibration dataset.`

And after this, the process ends up with a message of process being "Killed".

```
[VAIQ_NOTE]: Mem status(total mem: 125.38G, available mem: 107.53G).
 63%|██████████████████████████████████████████████████████████████████████████████████████████████████▊                                                          | 3222/5120 [22:33<1:27:07,  2.75s/it]
running_quantization_scripts/ft_quant/01-run_ft-quantization_torchvision_08_trained_weights_calib.sh: line 22: 28209 Killed                  python quantize_torchvision_v08_headref_trained_weights.py --dset_dir "/media/VA/databases/COMPORT/20220919/to_annotate" --ann_file "/media/VA/databases/COMPORT/20220919/to_annotate/comport_20220919_train.json" --float_model "output_trained/comport_20220919_train_0631/model_retinanet_17_best.pth" --quant_output "./quantization_results_comp_trained_vai20" --quant_mode calib --subset_len 200 --batch_size 2 --num_classes 6 --fast_finetune --ft_dir "/media/VA/databases/DRONE_VS_BIRD/images" --ft_file "/media/VA/databases/DRONE_VS_BIRD/annotations/DroneVsBird_train_clean.json"
```

So, I decided to reduce the number of images for the fast-finetuning process. I used a dataset of 1536 images for fast-finetuning. This time, the fast-finetuning process seems to run without errors (although it takes several hours to complete). However, the accurary loss I see at the end is greater than the one with direct quantization. 

```
[VAIQ_NOTE]: Loading NNDCT kernels...                                                                                                                                                                   

[VAIQ_NOTE]: Quantization calibration process start up...
                
[VAIQ_NOTE]: =>Quant Module is in 'cuda'.

[VAIQ_NOTE]: =>Parsing RetinaNet_mod...

[VAIQ_NOTE]: Start to trace model...

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 204/204 [00:00<00:00, 2204.58it/s, OpInfo: name = RetinaNet_mod/RetinaNetHead_mod[head]/RetinaNetRegressionHead_mod[regression_head]/10926, typ

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(./quantization_results_comp_trained_vai20/RetinaNet_mod.py)

[VAIQ_NOTE]: =>Get module with quantization.
2022-12-20 07:21:06.377 | DEBUG    | __main__:inference_quant_model:171 - Inference_quant on ../../../data/DRONE_VS_BIRD/images/dji_mavick_mountain_cruise/dji_mavick_mountain_cruise_0.png, in device c
uda:0

[VAIQ_NOTE]: =>Preparing data for fast finetuning module parameters ...

[VAIQ_NOTE]: Mem status(total mem: 125.38G, available mem: 117.79G).

[VAIQ_NOTE]: Memory cost by fast finetuning is 60.13 G.

[VAIQ_NOTE]: =>Find initial quantization steps for fast finetuning...
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1536/1536 [07:42<00:00,  3.32it/s]

[VAIQ_NOTE]: =>Fast finetuning module parameters for better quantization accuracy...
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 66/66 [13:33:39<00:00, 739.69s/it]
%%%%%%%%%%%%%%%%% final opt net loss:[0.5502090553442155, 0.00286597273727724, 0.012999053637493, 0.012568182480232235, 0.00821176284508884, 0.008272341218763945]

[VAIQ_NOTE]: =>Export fast finetuned parameters ...

[VAIQ_NOTE]: =>Exporting quant model parameters.(./quantization_results_comp_trained_vai20/param.pth)
2022-12-20 21:15:39.895 | INFO     | __main__:quantize:555 - Evaluation

Average Precision  (AP) @[ IoU=0.50      | area=   all | maxDets=100 ] = 0.231
```

I expected the fast-finetuning method to have less accuracy loss than the direct quantization one. However, it happened the other way around: I've got AP=0.410 in direct quantization, and AP=0.231 in fast-finetuning quantization.

Do anyone have a hint about what could be going wrong with this?
May have it been because of using less than 5120 images for fast-finetuning? But, how could I use 5120 images without having memory issues?

jimheaton, 2023-01-04:
What batch size are you using? Reducing that may help with the memory issue.

I have seen that fast fine tuning does not always result in better accuracy.

You will probably need to use QAT to improve the accuracy, this will especially be the case if your model uses DepthWise Convolutions.

daniperfer, 2023-01-09:
Hi, @jimheaton:
I have been using a batch size of 2 for quantizing my model. I will try a batch size of 1 to see if it behaves the same way.
There are not DepthWiseConv layers in my model, just Conv2d, BatchNorm2d and ReLUs.

I haven't got in touch with QAT so far. So, I will take a look at it...

kewang-xlnx, 2023-01-12:
Hi, @jimheaton:
Could you share your model and quantization scripts? 
Could I just need to use the origin torchvision RetinaNet to debug this issue?

daniperfer, 2023-01-18:
Hi, @kewang-xlnx:

I could not use the original torchvision RetinaNet model, because it was not directly quantizable (at least, in Vitis-AI 2.0 which is the version I have been using for this).
So, I had to make some modifications to the original torchvision RetinaNet code in order to be able to quantize it.

Although the code in my scripts may be a bit messy, I think I could share it with you so as to see how I am trying to apply the fast finetuning quantization. However, the files I would need to share occupy ~140 MB, and I cannot share them directly through GitHub. 
If you give me an email address, I could share a Drive download link with the email address.

By the way, what do you think about Quantization Aware Training? 
I've never tried it so far. But I am experiencing accuracy losses with direct quantization and fast fine-tuning quantization, so I wonder how difficult it might be to apply QAT...

kewang-xlnx, 2023-02-06:
Hi, @daniperfer:
My email address is kewang@xilinx.com.
You may refer to https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_quantizer/vai_q_pytorch/README.md for QAT.

daniperfer, 2023-02-07:
Hi, @kewang-xlnx:
I've shared with your email a link to a zip file in drive with the scripts I have been using for direct and fast-finetuning quantization in Vitis-AI, in case you want to take a look at them.
Thank you.

IkrameBeggar, 2023-09-03:
> Hi, @kewang-xlnx: I've shared with your email a link to a zip file in drive with the scripts I have been using for direct and fast-finetuning quantization in Vitis-AI, in case you want to take a look at them. Thank you.

hello, I want to use QAT for yolov5  using Vitis AI, can you share with me the files you have please. My email address is: ikrame.bgr@gmail.com
Thank you in advance