Xilinx/Vitis-AI #1124: xclbin.ex.run_summary not generated - vaitrace

anirudhitagi, 2023-01-05:
Hi,

I am using a ZCU104 with Vitis AI 2.6 and PYNQ v3.0.1 . 

I am trying to follow the [vaitrace](https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Starting-a-Simple-Trace-with-vaitrace) instructions for the [resent50_mt_py](https://github.com/Xilinx/Vitis-AI/tree/v2.5/examples/VART/resnet50_mt_py). executed the same instructions as given in the document: `python3 -m vaitrace_py ./resnet50.py 2 /usr/share/vitis_ai_library/models/resnet50/resnet50.xmodel` 
Although this appears to have run successfully, it did not generate the  **xclbin.ex.run_summary**, instead it generated the following:
[image]
it generates the **xrt.run_summary**.

I tried to read this file with vitis_analyzer 2021.1.1 and while it does read some info it is missing other information such as kernel data transfers, etc ...

Additionally I ccan confirm that the xmodel and .bit files were generated correctly, because I am able to run inference on it and that works as intended....

Please advise how to resolve this?

-AI

jileilxinlix, 2023-01-30:
@anirudhitagi ，
Thank you for your question, 
Q1: Can not find xclbin.ex.run_summary
From VERSION XRT2021.1 we changed xclbin.ex.run_summary file name to xrt.run_summary.

Q2: KERNAL DATA TRANSFTERS
Kernel data transfer profiling requires re-build bitstram using v++ with --profile options., detail refer link: https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/profile-Options

Additionally if you want to get more helpful info without rebuild bitstream, please add xrt_trace=true to xrt.ini. and details refer:https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/xrt.ini-File 

Jilei

qianglin-xlnx, 2023-04-22:
Closing since no activity for more than 2 months, please reopen if you still have any questions, thanks

closed 2023-04-22 by qianglin-xlnx