Xilinx/Vitis-AI #1126: Is it possible to compile pytorch fp32 model without quantize it?

lecoan, 2023-01-07:
`vai_c_xir` command uses xmodel as input, but it seems that we have to use `torch_quantizer.export_xmodel` to convert pytorch model into xmodel. However, I want to skip the quantization step and directly compile fp32 model, is it possible?

huisunCompiler, 2023-01-12:
Hi @lecoan,
The quantization step inserts "fix" ops before and after computing ops (e.g., conv2d, pool2d, etc.), so that the quantized ops can be launched on DPU. DPU only support "xxx-fix" ops (e.g., conv2d-fix, pool-fix, etc.). 
If you use xmodel without quantization step, the fp32 xmodel can also be compiled but may not be launched on DPU.

vccheng2001, 2023-01-26:
Hi, 
Can you elaborate on how the fp32 xmodel can be generated without using the `torch_quantizer`? I though the torch_quantizer requires you to quantize the model .

huisunCompiler, 2023-01-29:
Hi @vccheng2001 , 
Your thought is right. We **must** quantize the model through `torch_quantizer` to make the model run on DPU.

To generate a fp32 xmodel without using the `torch_quantizer`, you can use xir python interface to generate a xmodel directly. (I don't recommend do this, because the ops will not be partitioned to DPU.)
```python
import xir
import numpy as np
g = xir.Graph("toy")
data = g.create_op("data1", "data", {"data_type": "FLOAT32", "shape": [1, 48, 48, 3]}) # NHWC layout
weight = g.create_const_op("weight", np.random.randn(32, 3, 3, 3).astype(np.float32))
conv = g.create_op("conv", "conv2d",
                      {"kernel": [3, 3], "dilation": [1, 1], "stride": [1, 1], "pad": [1, 1, 1, 1], "pad_mode": "FLOOR"},
                      {"input": [data], "weights": [weight]})
g.serialize("toy.xmodel")
```

vccheng2001, 2023-01-30:
Thank you for the clarification!

closed 2023-02-11 by lecoan