Xilinx/Vitis-AI #1130: bert downloaded an empty tarball

jia0511, 2023-01-16:
https://www.xilinx.com/bin/public/openDownload?filename=pt_bert-tiny_SQuADv1.1_384_453M_3.0.zip
https://github.com/Xilinx/Vitis-AI/blob/master/model_zoo/model-list/tf_bert-base_SQuAD_128_22.34G_3.0/model.yaml
https://github.com/Xilinx/Vitis-AI/blob/master/model_zoo/model-list/pt_bert-large_%20SQuADv1.1_384_246.42G_3.0/model.yaml

wangxd-xlnx, 2023-01-16:
Hi @jia0511 

Thanks for your feedback. We will fix these download links as soon as possible.

wangxd-xlnx, 2023-01-17:
Fixed. @jia0511 You could have another try.

jzymessi, 2023-01-17:
Hi @wangxd-xlnx :
     I want to deploy bert model on VCK5000. 
     Is there any corresponding deployment code or deployment process?

wangxd-xlnx, 2023-01-17:
Hi @jzymessi 

The bert series models based on Gemm compiler deployed on VCK5000 will be released in Vitis-AI 3.0.1. It will take ~1 month, please stay tuned.

wangxd-xlnx, 2023-01-17:
By the way, there is a demo you could refer to. Though it's from Vitis-AI 2.5, not the latest version. @jzymessi

https://github.com/Xilinx/Vitis-AI/tree/2.5/examples/Transformer

jzymessi, 2023-01-17:
@wangxd-xlnx ：
Is bert deployment of vitis-ai version 3.0.1 called as xmodel?

closed 2024-03-19 by jia0511