Xilinx/Vitis-AI #1131: Vitis-AI 3.0 failed to run vaitrace tool

jzymessi, 2023-01-17:
I encountered the following problems when using vaitrace tool on VCK5000:
[image]

Here are my implementation steps:
step1:
add   -v /sys/kernel/debug:/sys/kernel/debug  --privileged=true \   in docker_run.sh
[image]
step2:
run docker 
[image]
step3:
sudo bash
[image]
but user is not root.
step4:
run vaitrace
[image]

How to solve the above problem? Thanks.

Dexter706, 2023-03-29:
Any update on this issue...I am also facing the same

Dexter706, 2023-04-02:
Hi @jzymessi 

Thanks for the reply.But for me it's still not working.XRT  is throwing the error.

`
vitis-ai-user@XXXX:/workspace/examples/vai_library/samples/classification$ sudo vaitrace ./test_jpeg_classification resnet50_pt sample_classification.jpg
INFO:root:VART will run xmodel in [NORMAL] mode
INFO:root:Executable file: /workspace/examples/vai_library/samples/classification/test_jpeg_classification
Analyzing symbol tables...
207 / 207
81 / 81
6 / 6
45 / 45
3 / 3
terminate called after throwing an instance of 'std::runtime_error'
  what():  XILINX_XRT not set
INFO:root:Generating VTF
ERROR:root:Xmodel info raw data empty
INFO:root:[vart_trace.csv]: Events number: 0

vitis-ai-user@XXXX:/workspace/examples/vai_library/samples/classification$ echo $XILINX_XRT
/opt/xilinx/xrt`

SuZhan0322, 2023-04-13:
Hi @jzymessi , Can you skip step3 and retry the vaitrace tool? Thanks!

jzymessi, 2023-04-13:
@Dexter706 
run follow:
vaitrace --fine_grained ./test_jpeg_classification xcompiler_model/reset50_pt.xmodel sample_classification

Irelia1, 2024-01-14:
sudo -E