Xilinx/Vitis-AI #1133: Using OFA to prune the cifar10 example encountered a problem

jia0511, 2023-01-17:
source 1_train_ofa_model.sh

Traceback (most recent call last):
  File "mynet_ofa_pruning.py", line 358, in <module>
    ofa_model = ofa_pruner.ofa_model(args.expand_ratio, args.channel_divisible, args.excludes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-optimizer_pytorch/lib/python3.7/site-packages/pytorch_nndct/pruning/ofa.py", line 168, in ofa_model
    mod_util.replace_modules(model, name, dynamic_modules, copy_ckpt=True)
  File "/opt/vitis_ai/conda/envs/vitis-ai-optimizer_pytorch/lib/python3.7/site-packages/pytorch_nndct/utils/module_util.py", line 155, in replace_modules
    copy_module_weights(modules[i], new_modules[i])
  File "/opt/vitis_ai/conda/envs/vitis-ai-optimizer_pytorch/lib/python3.7/site-packages/pytorch_nndct/utils/module_util.py", line 80, in copy_module_weights
    dst_module.conv.weight.data.copy_(src_module.weight.data)
RuntimeError: The size of tensor a (4) must match the size of tensor b (3) at non-singleton dimension 1

jia0511, 2023-01-17:
[**vitis-ai3.0**](https://github.com/Xilinx/Vitis-AI/tree/master/examples/vai_optimizer/pytorch/cifar10_ofa_pruning)

jia0511, 2023-02-07:
There are some updates?

DehuaTang, 2023-02-14:
Sorry, I didn't reproduce this issue.