Xilinx/Vitis-AI #1137: Vitis AI 2.0 Failed to Build Docker Image for GPU

shimoshida, 2023-01-18:
Now ` nvidia/cuda:11.0-cudnn8-runtime-ubuntu18.04` does not exist in Dockerhub.
```
Sending build context to Docker daemon   76.8kB
Step 1/61 : FROM nvidia/cuda:11.0-cudnn8-runtime-ubuntu18.04
manifest for nvidia/cuda:11.0-cudnn8-runtime-ubuntu18.04 not found: manifest unknown: manifest unknown
```
current available one is `nvidia/cuda:11.1.1-cudnn8-devel-ubuntu18.04`.

janifer112x, 2023-01-29:
@shimoshida, we've updated the base image to use label. nvidia/cuda:11.0.3-cudnn8-runtime-ubuntu18.04.  please try again and let us know if it works for you

shimoshida, 2023-01-29:
@janifer112x Oh, I have created PR: https://github.com/Xilinx/Vitis-AI/pull/1138. 
Should I delete this PR?

janifer112x, 2023-01-29:
@shimoshida , yes. please close that PR. if the latest commit has fixed your problem, I will close this issue,:)

closed 2023-01-29 by shimoshida