Xilinx/Vitis-AI #1139: Vitis-ai3.0

jia0511, 2023-01-19:
Hello xilinx,

It can be seen from the title of the document that it can be converted to fp16, but the content description is converted from fp16 to int8, so I want to confirm that the h5 model here is already an fp16 model. Currently, the quantizers of pytorch and tensorflow still cannot be converted to fp16 Format?

[image]

jia0511, 2023-02-07:
There are some updates?

renjiangyong, 2023-08-23:
you can try the vitis-3.5 version.@jia0511

closed 2023-08-25 by jia0511

zhenzhen-AMD, 2023-10-11:
Hi @jia0511 ，
The h5 model in the sample code is in float32 format. The example depicts its conversion to float16.