Xilinx/Vitis-AI #1140: Bert QAT metrics

LorenzoSun-V, 2023-01-19:
Thank you for your hard work to realease VitisAI 3.0 ! :blush: :rocket::rocket::rocket:
I'm working on realeased [Bert-based](https://github.com/Xilinx/Vitis-AI/tree/master/model_zoo/model-list/pt_bert-base_SQuADv1.1_384_70.66G_3.0) project and run the whole flow successfully.

However, I found metrics of QAT model are much lower than FP32 model:
```
FP32 metrics:
{
    "epoch": 2.0,
    "eval_exact_match": 81.0406811731315,
    "eval_f1": 88.3364950325187,
    "eval_samples": 10784
}

QAT metrics:
{
    "epoch": 3.0,
    "eval_exact_match": 62.07190160832545,
    "eval_f1": 73.44366320006854,
    "eval_samples": 10784
}
```

In addition, the parameter ` --nndct_ip_v70_bert_qat True`  in QAT script is confusing indeed. I could not find any matched parameter in QAT code.  Would you mind tell me the usage of these parameters?(Maybe writing comments in code or README is a nice way next time.)

The last but not least, the code only export the onnx model in deploy phase but no xmodel. Is it not yet supported to deploy xmodel on board?

Alexiazzf, 2023-02-06:
Hi. We will officially release BERT models at VitisAI 3.0.1. So some parameters in BERT project did not sync to vai_q_pytorch yet. If you want a good result, you can add a new parameter  `--nndct_ip_v70_bert True` after the parameter  `--nndct_ip_v70_bert_qat True`. These two parameters set specific activation quantization methods which are directly recgnized by vai_q_pytorch.

In addition, BERT project is not used for xcompiler in this version, so it is not supported to deploy xmodel yet.

closed 2023-03-02 by niuxjxlnx

LorenzoSun-V, 2023-03-03:
Thanks for your reply. 
Does the BERT model only support deployment on v70? Do you have any plan to deploy BERT on VCK5000?

Alexiazzf, 2023-03-06:
Hi @LorenzoSun-V ,

BERT models only support on v70 for now. We do not plan to deploy them on VCK5000 yet.