Xilinx/Vitis-AI #1141: Is Graph_runner mode suitable for running models with both DPU and CPU operations?

AdrFebles, 2023-01-19:
Hi!
I have a xmodel with DPU operations followed by some transpose and fix2floats, you can see a part of the scheme bellow:
[image]

I used "graph_runner" mode to execute all the operations compiled in the .xmodel including both DPU and CPU operations on the ZCU102. I expected to see an increase in the frames per second ratio compared to what was obtained with the "runner" mode, which only runs DPU operations so the CPU ones needed to be implemented in the app.
Following are the performance results in each case:

**Graph_runner** mode: (FPS=0.61, total frames = 100.00 , time=164.941214 seconds)

**Runner** mode: (FPS=5.32, total frames = 100.00 , time=18.801671 seconds)

@Shreyas-NR i don't know if you tried with both modes?
Can anyone help me understand how graph_runner works and why it takes so long to inference my model?

Thanks!

lishixlnx, 2023-01-25:
That's correct running result.
It's because   there are so many cpu op in the graph.
You can see if using graph-runner mode, all the cpu-op should be executed, and some of those can be optimized if you write your own implementation.

qianglin-xlnx, 2023-04-22:
Closing since no activity for more than 2 months, please reopen if you still have any questions, thanks

closed 2023-04-22 by qianglin-xlnx