Xilinx/Vitis-AI #1142: Kria KV260 Setup Documentation Update Schedule

osamuraislice, 2023-01-20:
Thanks for releasing Vitis-AI V3.0. I will start development using this repository.🎉

I understood that the Vitis-AI/setup folder in the V2.5 branch has been moved to the Vitis-AI/board_setup folder.
[Vitis-AI/setup](https://github.com/Xilinx/Vitis-AI/tree/master/board_setup)

But, the links to the workflow and board setup documentation at the bottom of this page are broken.

I would like to use Vitis-AI V3.0 with Kria KV260, are there any plans to update the documentation?

qianglin-xlnx, 2023-01-24:
Hi @osamuraislice 
Please refer to : https://xilinx.github.io/Vitis-AI/docs/board_setup/board_setup_mpsoc.html

osamuraislice, 2023-01-25:
Thank you for your response.
I will check the [github.io](https://xilinx.github.io/Vitis-AI/docs/board_setup/board_setup_mpsoc.html) link and start developing on KV260.
I will close this issue now that the issue is resolved.

closed 2023-01-25 by osamuraislice