Xilinx/Vitis-AI #1145: Can compiled models in Vitis AI 3.0 be executed on VART 2.5 (petalinux 22.1) on ZCU104?

dextroza, 2023-01-24:
Hi,
I have compiled model in Vitis AI 3.0 and tried to run it on VART 2.5 but I got the error:

"dpu_runner_base_imp.cpp:676] CHECK fingerprint fail ! model_fingerprint 0x101000056010407 dpu_fingerprint 0x101000016010407 
F0123 14:10:57.191071  1500 dpu_runner_base_imp.cpp:648] fingerprint check failure."

Am I missing something or this is just not compatible? Another idea is only to prune model in Vitis AI 3.0 and then quantize and compile in Vitis AI 2.5 in order to run on VART 2.5. Should we try this way or we should not mix with Vitis AI dockers and VARTs? 

Best regards,
dextroza

qianglin-xlnx, 2023-01-24:
Hi @dextroza 
It's recommended to recompile the model with the corresponding dpu arch.json, as shown below. You can use VAI2.5 or VAI3.0 docker to compile the model.
The new arch.json.
```
{
"fingerprint": "0x101000016010407"
}
```

dextroza, 2023-01-25:
I also tried that dpu_arch.json in Vitis AI 3.0:
`{
"fingerprint": "0x101000016010407"
}`

but there was an error in compiling: "ERROR: NO FRONT END SPECIFIED"

Good news: our quantized model in Vitis AI 3.0 can be compiled in Vitis AI 2.5 and it works on VART 2.5 (with "fingerprint": "0x101000016010407")

xihuazhanghui, 2023-01-30:
Hi, @dextroza 
You can set the environment XLNX_ENABLE_FINGERPRINT_CHECK=0 to ignore fingerprint check, The 3.0 have a little change on dpu ip have a new fingerprint, but the functionality is same with 2.5

xihuazhanghui, 2023-01-30:
> 
Can you show me your compile command and which docker did you use?

closed 2023-02-03 by dextroza

Qiang-Feng98, 2023-08-11:
Hi, @xihuazhanghui
 I seem to have encountered a similar error:
**_W0811 00:39:09.906442 95090 dpu_runner_base_imp.cpp:676] CHECK fingerprint fail ! model_fingerprint 0x101000056010407 dpu_fingerprint 0x101000016010407
F0811 00:39:09.906534 95090 dpu_runner_base_imp.cpp:648] fingerprint check failure.
*** Check failure stack trace: ***
Aborted_**
I checked my vart version and it is 2.5.
[image]
So do I have to compile my model under Vitis AI 2.5?
 Can I update the image of my target board (KV260) (or update the vart version)?
Thanks very much.

quentonh, 2023-08-11:
With respect to Fingerprint and bitstream compatibility, the version of the MPSoC IP that was released in v3.0 and 3.5 is a different version than was released in v2.5.  FWIW, the IP versions are documented here:  https://xilinx.github.io/Vitis-AI/3.5/html/docs/reference/version_compatibility.html#all-releases

Here is an example of how to interpret the Fingerprint when you receive this type of message.

model_fingerprint 0x101000056010407 dpu_fingerprint 0x101000016010407

First, we can confirm that both Fingerprints are of DPU Type = 1, which indicates the DPUCZ architecture, see [here](https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_runtime/target_factory/src/target_factory.cpp#L123).  Next, we can see that in the above example, the model was compiled targeting a DPU ISA1 architecture [(48th bit = 1)](https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_runtime/target_factory/src/target_factory.cpp#L172), and that the DPU in the bitstream is also ISA1 architecture (48th bit = 1).

However, what is also clear is that the feature [codes](https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_runtime/target_factory/src/target_factory.cpp#L171) (56010407 & 16010407) in this example do not match.  The “feature code” breakdown relevant to this thread can be found in the respective prototxt file for the DPU in "target factory).  

2.5 release:
DPUCZDX8G_ISA1_B4096 = 0x16010407
See [here](https://github.com/Xilinx/Vitis-AI/blob/2.5/src/Vitis-AI-Runtime/VART/target_factory/targets/DPUCZDX8G_ISA1_B4096.prototxt#L18) 

3.0 release:
DPUCZDX8G_ISA1_B3136_MAX_BG2 = 0x56010407
See [here](https://github.com/Xilinx/Vitis-AI/blob/3.0/src/vai_runtime/target_factory/targets/DPUCZDX8G_ISA1_B4096.prototxt#L18)

The implication here is that you have a Vitis AI 3.0 compiled model and are attempting to deploy it on an older DPU architecture, and if you circle back to the first link in this email, you will see that the IP was revised between the 2.5 and 3.0 releases.

Does this help?