Xilinx/Vitis-AI #1146: Pytorch padding fusion broke Vitis AI 3.0

ash8327, 2023-01-26:
Hi,

We had a model that includes ZeroPad2d, Conv2d, BatchNorm2d, ReLU fusion that for Vitis AI 2.5 would compile and run in the DPU. With the only change being an update to Vitis AI 3.0 the ZeroPad2d is now ran on the CPU instead of the DPU. Was there a change to BatchNorm2d or anything that could cause this change? 

This is being ran on the Versal VCK190 development kit.

huisunCompiler, 2023-01-29:
Hi @ash8327 ,
The BatchNorm2d layer will not affect the partition result of the ZeroPad2d layer. Only Conv2d layer will influence the partition result of the ZeroPad2d.

The ZeroPad2d layer will be merged to Conv2d and attached as **the "paddings" attribute** in the Conv2d layer, only if ZeroPad2d has mode="CONSTANT", constant_values=0.0, paddings=[0, 0, x, x, x, x, 0, 0]. Otherwise, ZeroPad2d will **not** be merged to Conv2d layer and will be partitioned to CPU.

Could you please provide the attributes of the ZeroPad2d layer and the Conv2d layer or the xmodel before vai_c_xir. So that we can analysis the problem in detail and provide help to your question. If you are willing to share the attributes of the ZeroPad2d layer and Conv2d layer, the attributes can be seen through the [netron](https://github.com/lutzroeder/netron) app.

ash8327, 2023-01-31:
Hi @huisunCompiler ,

We are looking into this now. Was there any change made between 2.5 and 3.0 that required setting mode="CONSTANT" and not implicit? The code did utilize the DPU in 2.5.

huisunCompiler, 2023-02-01:
Hi @ash8327 ,
The compiler has made a change on merging pad condition. We set the `pad_mode` of the following Conv2d to "FLOOR" after the ZeroPad2d layer is merged. However, this change seems to have no influence on your case. 
I think I should ask my colleagues to answer your question.