Xilinx/Vitis-AI #1151: feature request: support bfloat16/float16 quantization in pytorch quantizer

rderp, 2023-02-02:
It seems like only the tensorflow quantizer supports floating point quantization. We'd like this to be made available in torch quantizer apis.

rderp, 2023-02-02:
Actually, perhaps I jumped the gun. Appears like there is a resnet and mobilenet torch model quantization example. Documentation seems to imply that only bfloat16 quantization is supported in tensorflow in UG1414. Perhaps a similar thing for torch should be added.

rderp, 2023-02-02:
Release notes say it is supported for release 3.0.

niuxjxlnx, 2023-02-06:
@rderp ,

**bfloat16** is not quantized in vitis-ai pytorch quantizer 3.0 version. It only supports **float16**. 
This is because that Pytorch support of **bfloat16** is not completed.

closed 2023-02-06 by niuxjxlnx

rderp, 2023-02-06:
@niuxjxlnx Is there an example of how to do float16 quantization with the SDK?

rderp, 2023-02-06:
@niuxjxlnx could you reopen this around float16 rather than bfloat16?

niuxjxlnx, 2023-02-07:
@rderp ，

It does not need any special setting or call any different function. You can quantize your float16 model by the approach of quantizing float32 model.