Xilinx/Vitis-AI #1152: How to get the int8 param?

JF-Lee, 2023-02-02:
I want to use this quantization platform to quantify Pointpillars and export the quantized parameters(int8 weights&bias). What should I do?

I downloaded the corresponding file from 
[](https://github.com/Xilinx/Vitis-AI/blob/master/model_zoo/model-list/pt_pointpillars_kitti_12000_100_11.2G_3.0/model.yaml)
files:
- name: pt_pointpillars_kitti_12000_100_11.2G_3.0
  type: float & quantized
  board: GPU
  download link: https://www.xilinx.com/bin/public/openDownload?filename=pt_pointpillars_kitti_12000_100_11.2G_3.0.zip
  checksum: 5ad7a30c21dc6c2041ff03c827fd53dc

However, .xmodel only has model topology and does not contain int8 parameters or quantization coefficients similar to DQD structure. 
I think qat_ converted.pth is the parameter obtained after QAT, and then the quantization coefficient is required to obtain the corresponding int8 parameter.

niuxjxlnx, 2023-02-06:
@JF-Lee ,

After quantization, the parameters stored everywhere are all float values equivalent to (quantization scale * int8).
So you can get int8 values by (float value / quantization scale).

JF-Lee, 2023-02-06:
> @JF-Lee ,
> 
> After quantization, the parameters stored everywhere are all float values equivalent to (quantization scale * int8). So you can get int8 values by (float value / quantization scale).

Thanks for you reply! quant_info.json contain [bitwidth, fp](scale=2^fp) which to convert float param to int8 param.

I download the .xmodel from 
- name: pointpillars_kitti_12000_1_pt
  type: xmodel
  board: zcu102 & zcu104 & kv260
  download link: https://www.xilinx.com/bin/public/openDownload?filename=pointpillars_kitti_12000_1_pt-zcu102_zcu104_kv260-r3.0.0.tar.gz

but I have some questions：
If the denotation means the fp(scale=2^fp)? If so, why the first conv layer has the different fp between quant_info.json and this .xmodel which has been complied. 
[image]

torukskywalker, 2024-03-19:
How to get the value "quantization scale" in python code after quantization ?