Xilinx/Vitis-AI #1154: multi_task_v3_pt not available in v3.0

maciekbaczmanski, 2023-02-05:
Hi, 
As far as I can see from the documentation, it is said that the multi_task_v3_pt model is supported by Vitis-AI v3.0, but it isn't available in the model zoo library. Are there any plans to add it? Also, the 2.5 version contains multiple bugs (visible when running the demo) and instruction about dataset preparation is quite unclear as it mentions .py files which aren't available in the repository. Any chance to look at dataset preparation instructions too? 
Thanks!

wangxd-xlnx, 2023-02-17:
Hi @maciekbaczmanski 

Thank you for feedback. This model is indeed not in the Vitis-AI 3.0 model list because of some problems with the dataset.
If you really want to use this model, you may need to rely on version 2.5.

maciekbaczmanski, 2023-02-20:
Hi @wangxd-xlnx , ok, thanks. The issue I have with files for 2.5 version is that there are two python scripts mentioned in readme:
[image]
`bdd_to_yolo.py` and `./data/labels.py` are missing in the file tree. could you help with finding these files?
I've downloaded the model repository from https://www.xilinx.com/bin/public/openDownload?filename=pt_multitaskv3_mixed_320_512_25.44G_2.5.zip