Xilinx/Vitis-AI #1155: how to import torchaudio?

anirudhitagi, 2023-02-05:
Greetings,

I intend to run the [Emformer](https://pytorch.org/audio/main/tutorials/online_asr_tutorial.html) using the Vitis-AI 2.5 on an Ultra96-V2.

But I am unable to import the torchaudio library. Here is the error I am facing:
````
(vitis-ai-pytorch) Vitis-AI /workspace > pip install torchaudio
Collecting torchaudio
  Downloading torchaudio-0.13.1-cp37-cp37m-manylinux1_x86_64.whl (4.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.2/4.2 MB 12.8 MB/s eta 0:00:00
Collecting torch==1.13.1
  Downloading torch-1.13.1-cp37-cp37m-manylinux1_x86_64.whl (887.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 887.5/887.5 MB 5.0 MB/s eta 0:00:00
Collecting nvidia-cuda-runtime-cu11==11.7.99
  Downloading nvidia_cuda_runtime_cu11-11.7.99-py3-none-manylinux1_x86_64.whl (849 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 849.3/849.3 kB 18.2 MB/s eta 0:00:00
Collecting nvidia-cuda-nvrtc-cu11==11.7.99
  Downloading nvidia_cuda_nvrtc_cu11-11.7.99-2-py3-none-manylinux1_x86_64.whl (21.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 21.0/21.0 MB 17.1 MB/s eta 0:00:00
Requirement already satisfied: typing-extensions in /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages (from torch==1.13.1->torchaudio) (4.2.0)
Collecting nvidia-cudnn-cu11==8.5.0.96
  Downloading nvidia_cudnn_cu11-8.5.0.96-2-py3-none-manylinux1_x86_64.whl (557.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 557.1/557.1 MB 2.3 MB/s eta 0:00:00
Collecting nvidia-cublas-cu11==11.10.3.66
  Downloading nvidia_cublas_cu11-11.10.3.66-py3-none-manylinux1_x86_64.whl (317.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 317.1/317.1 MB 9.0 MB/s eta 0:00:00
Requirement already satisfied: setuptools in /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages (from nvidia-cublas-cu11==11.10.3.66->torch==1.13.1->torchaudio) (59.8.0)
Requirement already satisfied: wheel in /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages (from nvidia-cublas-cu11==11.10.3.66->torch==1.13.1->torchaudio) (0.37.1)
Installing collected packages: nvidia-cuda-runtime-cu11, nvidia-cuda-nvrtc-cu11, nvidia-cublas-cu11, nvidia-cudnn-cu11, torch, torchaudio
  Attempting uninstall: torch
    Found existing installation: torch 1.10.1
    Uninstalling torch-1.10.1:
      Successfully uninstalled torch-1.10.1
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
torchvision 0.11.2+cpu requires torch==1.10.1, but you have torch 1.13.1 which is incompatible.
Successfully installed nvidia-cublas-cu11-11.10.3.66 nvidia-cuda-nvrtc-cu11-11.7.99 nvidia-cuda-runtime-cu11-11.7.99 nvidia-cudnn-cu11-8.5.0.96 torch-1.13.1 torchaudio-0.13.1
````

After this when I try to check if the libraries are instaleld correctly I see the following error:
```

(vitis-ai-pytorch) Vitis-AI /workspace > python
Python 3.7.12 | packaged by conda-forge | (default, Oct 26 2021, 06:08:53) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torchaudio
>>> import torch
>>> from pytorch_nndct.apis import torch_quantizer
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'
Segmentation fault (core dumped)
```

Please advise how to resolve this?

niuxjxlnx, 2023-02-16:
@anirudhitagi ,

Do you notice the following log? When you install torchaudio, you uninstalled torch 1.10.1 in docker. 
Please keep torch installed in docker unchanged when you install other python modules. 
You can choose compatible version of torchaudio to install per this page https://pytorch.org/get-started/previous-versions/

```
Installing collected packages: nvidia-cuda-runtime-cu11, nvidia-cuda-nvrtc-cu11, nvidia-cublas-cu11, nvidia-cudnn-cu11, torch, torchaudio
  Attempting uninstall: torch
    Found existing installation: torch 1.10.1
    Uninstalling torch-1.10.1:
      Successfully uninstalled torch-1.10.1
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
torchvision 0.11.2+cpu requires torch==1.10.1, but you have torch 1.13.1 which is incompatible.
```

anirudhitagi, 2023-02-16:
Hi @niuxjxlnx ,
Thank you for the detailed explanation... 
But the model I am interested in requires ([Emformer](https://pytorch.org/audio/main/tutorials/online_asr_tutorial.html)) torchaudio==0.13.1, which in turn requires torch==1.13.1......
are these versions not compatible with the Vitis-AI 2.5?

anirudhitagi, 2023-02-16:
Please my above comment. it looks to have worked.  - https://github.com/Xilinx/Vitis-AI/issues/1170

closed 2023-02-16 by anirudhitagi