Xilinx/Vitis-AI #1159: Vitis AI 2.5: API_0 yolov3_adas_pruned_0_9 VART/adas_detection slower than API_1 yolov3_adas_pruned_0_9 Vitis-AI-Library/yolov3

omgitsmj24, 2023-02-07:
Despite using the same xmodel file, these two API_0: VART-adas_detection and API_1: Vitis-AI-Library-yolov3 show massive difference in FPS and speed during inference. I'm currently trying to implement my own model using VART, but it also showed that API_0 VART is much slower compared to API_1 and 2. Is this intended or there have been some bad configuration on my end?

API_0: VART - yolov3_adas_pruned_0_9
[image]

API_1: Vitis-AI-Library - yolov3_adas_pruned_0_9
[image]

DPU: DPUCZDX8G_ISA1_B4096
BOARD: ZCU102
Vitis-AI version: 2.5.0

guohaot-Xlnx, 2023-02-07:
Hi, @omgitsmj24 
Thank you for your feedback.
I tested 2 samples from vart and Vitis-AI-Library.
I preliminarily found the pre-processing and post-processing speed of the vart to be slower than the Vitis-Ai-Library. The main reason is that we only optimized for Vitis-AI-Library. I recommend using Vitis-AI-Library's interface for deployment.

Regards

qianglin-xlnx, 2023-03-27:
Closing since no activity for more than 1 month, please reopen if you still have any questions, thanks