Xilinx/Vitis-AI #1160: Missing GPU support for Vitis AI Tensorflow2 Optimizer

acg93-pixel, 2023-02-07:
Hi, I'm using Vitis AI Docker built for gpu: both 2.5 and 3.0 versions. I've noticed that GPU is not utilized at all during pruning process. Going through source code I've came across commented lines where GPU is supposed to be initialized inside /opt/vitis_ai/conda/envs/vitis-ai-optimizer_tensorflow2/lib/python3.8/site-packages/tf_nndct/optimization/runner.py script.
It's very inconvenient an time consumig.

akuzminski, 2023-02-10:
I'm also having the same problem, annoying....

yuwang-xlnx, 2023-02-14:
Hi, @acg93-pixel and @akuzminski, 

This issue is due to the cpu version of TensorFlow installed in Optimizer docker.
A quick fix for this issue:
```
conda activate vitis-ai-optimizer_tensorflow2
conda install tensorflow=2.10.0=cuda112py38hded6998_0 -c conda-forge
```

ioanacretu97, 2023-03-21:
@acg93-pixel can you please tell me how did you download the  vitis-ai 3.0 version for gpu? I can find only the cpu one. Thank you

nhphuong91, 2023-08-29:
> Hi, @acg93-pixel and @akuzminski,
> 
> This issue is due to the cpu version of TensorFlow installed in Optimizer docker. A quick fix for this issue:
> 
> ```
> conda activate vitis-ai-optimizer_tensorflow2
> conda install tensorflow=2.10.0=cuda112py38hded6998_0 -c conda-forge
> ```

I applied this fix but it didn't work. Result of `conda list` after using your cmd is as below:
```bash
.
tensorflow                2.8.0           cpu_py38h66f0ec1_0    conda-forge
tensorflow-base           2.8.0           cpu_py38hdf8f09a_0    conda-forge
tensorflow-estimator      2.8.0           cpu_py38h4e23bc6_0    conda-forge
.
vai_optimizer_tensorflow2_gpu 2.5.0            py38h415e102_7    file:///scratch/conda-channel
.
```
It gave a very long installation log but I notice this line at the end of it:
```bash
Your installed version is: 0
```
Would you have any idea how to fix it?

wangywy, 2023-08-29:
Hi @nhphuong91,

The specified package exists in conda-forge channel, please check if your conda is configured correctly:
https://anaconda.org/conda-forge/tensorflow/files?version=2.10.0

The issue has been fixed in Vitis AI 3.5, you can use it to avoid this issue.

nhphuong91, 2023-08-29:
> Hi @nhphuong91,
> 
> The specified package exists in conda-forge channel, please check if your conda is configured correctly: https://anaconda.org/conda-forge/tensorflow/files?version=2.10.0
> 
> The issue has been fixed in Vitis AI 3.5, you can use it to avoid this issue.

Hi, I'm currently stuck at Ubuntu 18.04. Which is not supported by VAI 3.5.
About the conda config, I believe it has been configured properly to get pkg from conda-forge. However, is there anyway to update its database or this process is done automatically?
For further information, here are the head & tail of installation log:
```bash
(vitis-ai-optimizer_tensorflow2) Vitis-AI /workspace > conda install tensorflow=2.10.0=cuda112py38hded6998_0 -c conda-forge
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: |
Found conflicts! Looking for incompatible packages.                                                                                                                 failed                                                                                                                                                                \

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package libnsl conflicts for:
scipy -> python[version='>=3.9,<3.10.0a0'] -> libnsl[version='>=2.0.0,<2.1.0a0']
.
```

```bash
.
Package platformdirs conflicts for:
scipy -> pooch -> platformdirs[version='>=2.5.0']
pooch -> platformdirs[version='>=2.5.0']
platformdirsThe following specifications were found to be incompatible with your system:

  - feature:/linux-64::__cuda==12.1=0
  - feature:/linux-64::__glibc==2.27=0
  - feature:/linux-64::__unix==0=0
  - feature:|@/linux-64::__cuda==12.1=0
  - feature:|@/linux-64::__glibc==2.27=0
  - feature:|@/linux-64::__unix==0=0
  - abseil-cpp -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - aiohttp -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - brotlipy -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - bzip2 -> libgcc-ng[version='>=9.3.0'] -> __glibc[version='>=2.17']
  - c-ares -> libgcc-ng[version='>=9.4.0'] -> __glibc[version='>=2.17']
  - cffi -> libgcc-ng[version='>=9.4.0'] -> __glibc[version='>=2.17']
  - click -> __unix
  - click -> __win
  - cryptography -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - cudatoolkit -> __glibc[version='>=2.17,<3.0.a0']
  - cudatoolkit -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - cudnn -> __glibc[version='>=2.17|>=2.17,<3.0.a0']
  - frozenlist -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - giflib -> libgcc-ng[version='>=7.5.0'] -> __glibc[version='>=2.17']
  - google-auth-oauthlib -> click[version='>=6.0.0'] -> __unix
  - google-auth-oauthlib -> click[version='>=6.0.0'] -> __win
  - grpc-cpp -> libgcc-ng[version='>=11.2.0'] -> __glibc[version='>=2.17']
  - grpcio -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - h5py -> libgcc-ng[version='>=9.4.0'] -> __glibc[version='>=2.17']
  - hdf5 -> libgfortran-ng -> __glibc[version='>=2.17']
  - icu -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - jpeg -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - keras -> tensorflow[version='>=2.2'] -> __cuda
  - keras -> tensorflow[version='>=2.2'] -> __glibc[version='>=2.17']
  - keyutils -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - krb5 -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - libaec -> libgcc-ng[version='>=9.4.0'] -> __glibc[version='>=2.17']
  - libblas -> libgcc-ng[version='>=9.3.0'] -> __glibc[version='>=2.17']
  - libcblas -> libgcc-ng[version='>=9.3.0'] -> __glibc[version='>=2.17']
  - libcurl -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
[... 26 lines cut ...]
  - snappy -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - sqlite -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - tensorboard -> libgcc-ng[version='>=7.3.0'] -> __glibc[version='>=2.17']
  - tensorboard-data-server -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - tensorflow-base -> __cuda
  - tensorflow-base -> __glibc[version='>=2.17']
  - tensorflow-base -> cudatoolkit[version='>=11.2,<12'] -> __glibc[version='>=2.17,<3.0.a0']
  - tensorflow-base -> nccl[version='>=2.14.3.1,<3.0a0'] -> __cuda[version='10.2|10.2.*|11.0|11.0.*|>=11.2,<12|11.1|11.1.*']
  - tensorflow-estimator -> __glibc[version='>=2.17']
  - tensorflow-estimator -> cudatoolkit[version='>=11.2,<12'] -> __glibc[version='>=2.17,<3.0.a0']
  - tensorflow-estimator -> tensorflow-base==2.12.1=cuda112py39h7e9974e_0 -> __cuda
  - tensorflow==2.10.0=cuda112py38hded6998_0 -> __cuda
  - tensorflow==2.10.0=cuda112py38hded6998_0 -> tensorflow-base==2.10.0=cuda112py38h6b2b66c_0 -> __glibc[version='>=2.17']
  - tk -> libgcc-ng[version='>=9.4.0'] -> __glibc[version='>=2.17']
  - urllib3 -> pysocks[version='>=1.5.6,<2.0,!=1.5.7'] -> __unix
  - urllib3 -> pysocks[version='>=1.5.6,<2.0,!=1.5.7'] -> __win
  - vai_optimizer_tensorflow2_gpu -> cudatoolkit[version='>=11.0.3,<11.1.0a0'] -> __glibc[version='>=2.17|>=2.17,<3.0.a0']
  - vai_optimizer_tensorflow2_gpu -> tensorflow=2.8.0 -> __cuda
  - wrapt -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - xz -> libgcc-ng[version='>=7.5.0'] -> __glibc[version='>=2.17']
  - yarl -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - zlib -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']

Your installed version is: 0
```

janifer112x, 2023-08-29:
Hi @nhphuong91,
maybe you could try with pip install after activate the conda env? 
pip install tensorflow-gpu==2.10.0

janifer112x, 2023-08-29:
@nhphuong91  as far as I remember optimizer tensorflow 2 was built with python3.8, so I'm confused by the comment.  I got the pip installation commanad from PYPI: https://pypi.org/project/tensorflow-gpu/2.10.0/