Xilinx/Vitis-AI #1163: Segmentation fault error running vart.Runner.create_runner in python script (Ubuntu 22.04)

mmarquez999, 2023-02-07:
Hi,

I'm trying to execute a python script inside a Docker container with image Ubuntu 22.04, on a ZCU102 board with Petalinux. The Docker image has installed the packages vitis-ai-library and vitis-ai-runtime and I'm able to import successfully the vart and xir modules in Python.

However, when the code reaches the line where I call the _vart.Runner.create_runner()_ method, it exists with Segmentation Fault (core dumped) error, without further explanation. The xmodel that the method receives has been compiled in vitis-ai 2.0, which matches the version of the vitis-ai-library and vitis-ai-runtime packages in the container. 

Is there anything that I'm missing??

Thanks in advance for your support!

mmarquez999, 2023-02-09:
Hi @qianglin-xlnx 

I tried to run the _xdputil query_ and _xdputil status_ commands directly on the ZCU102 board (on Petalinux) and then inside the Ubuntu Docker container running on the board. The first one was successfully but the execution inside the container was not. As you can see below, we get another "Segmentation fault (core dumped)" error. 
`root@f833092d31f2:/vart# xdputil status
/usr/bin/xdputil: line 20:    30 Segmentation fault      (core dumped) /usr/bin/python3 -m xdputil $*
root@f833092d31f2:/vart# xdputil query 
/usr/bin/xdputil: line 20:    32 Segmentation fault      (core dumped) /usr/bin/python3 -m xdputil $*
root@f833092d31f2:/vart# show_dpu
Segmentation fault (core dumped)`

I guess that the docker container does not have access to the DPU configuration located at the Petalinux host of the board. 

It seems that what I'm missing is how to give access to the docker container to the DPU drivers of the board. Would it make sense to mount the corresponding directories located at the board into the container so it could get the drivers and configurations from there? And in that case, what directories shoul I bring into the container?

Thanks for your help!!

xihuazhanghui, 2023-02-10:
The edge (ZCU102) board did not support run in docker container.

mmarquez999, 2023-02-13:
Hello @xihuazhanghui,

The ZCU102 does actually support running docker containers on it: as you can read above, I'm already running them without any problem. The issue here is how to use the DPU resource from inside, since it does not seem to be able to access it.

qianglin-xlnx, 2023-06-17:
This is not the use case that zcu102 to support.

closed 2023-06-17 by qianglin-xlnx

mmarquez999, 2023-06-19:
Just in case someone is trying to do the same, I managed to finally make it work.

sandyagur8, 2024-11-05:
Hello @mmarquez999 
I am hitting with the same issue on Kria KV260 starterkit board. Can you please help me with the steps you have taken to solve the issue?  I am running the accelerated application on  Ubuntu 22.04 LTS

dmesg |tail is 
```
[ 1295.636278] zocl-drm axi:zyxclmm_drm:  ffff00080a6d9c10 _kds_fini_client: Client pid(3249) has 1 opening context for 0 slot
[ 1295.636289] zocl-drm axi:zyxclmm_drm:  ffff00080a6d9c10 kds_del_context: Client pid(3249) del context Domain(0) CU(0x0)
[ 1295.636297] zocl-drm axi:zyxclmm_drm:  ffff00080a6d9c10 _kds_fini_client: Removing CU Domain[0] CU Index [0]
[ 1295.636307] [drm] bitstream 76ef2b72-faa6-a00c-5b37-640b5ef9670e unlocked, ref=0
[ 1295.636318] zocl-drm axi:zyxclmm_drm: zocl_destroy_client: client exits pid(3249)
[ 1295.636489] zocl-drm axi:zyxclmm_drm: zocl_destroy_client: client exits pid(3249)

```

sandyagur8, 2024-11-06:
@mmarquez999 
Thankyou for the response .
I am trying to run the accelerated application on the kria  board which has ubuntu 22.04 installed in it.

mmarquez999, 2024-11-07:
Mmm that's weird: if you read the whole thread of this issue, I did'nt have problems running the application on the board (it could access the DPU without any issue). My problem was when running the application inside a Docker container, that it could not access the DPU and I got a segmentation fault instead.

Have you tried to run the commands that I used of the xdputil tool to check the DPU information? As qianglin-xlnx told me, it's just to check the environment is ok in the board