Xilinx/Vitis-AI #1165: :bug: Quantization print bug

LorenzoSun-V, 2023-02-10:
When I QAT model in VitisAI3.0 env, I met a annoying print info:
[image]

I check the source code and found the `output_size` of `py_nndct.nn.AdaptiveAvgPool2d` has changed to `[1,1]` in 3.0 compared to `1` in 2.5:
[image]
[image]

Therefore, the value of expression which in the red box of the figure below is always `true` because `self.output_size` is `[1,1]`:
[image]

LorenzoSun-V, 2023-02-10:
```
if (isinstance(self.output_size,
                   (tuple, list)) and tuple(self.output_size) !=
        (1, 1)) or (isinstance(self.output_size, int) and self.output_size != 1):
      print(
          "[WARNING] For AdaptiveAvgPooling, DPU only supports output size=1"
      )
```
I suppose the code above may solve the print bug.

yuwang-xlnx, 2023-02-14:
> I check the source code and found the `output_size` of `py_nndct.nn.AdaptiveAvgPool2d` has changed to `[1,1]` in 3.0 compared to `1` in 2.5: [image] [image]
> 
Hi @LorenzoSun-V,

The scripts in above screenshots are used in PTQ, not QAT.
Can you confirm what the output_size of the Pooling layer is ? A simple way to do this is adding a print statement in https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/quantization/modules/pooling.py#L69:
`print(self.output_size)`

LorenzoSun-V, 2023-02-15:
Hi @yuwang-xlnx , I have added the `print` code in `pooling.py` and the output is like:
[image]

`self.output_size` is equal to `[1, 1] indeed.

yuwang-xlnx, 2023-02-15:
@LorenzoSun-V , Thanks for your feedfack . We'll fix this issue.

closed 2023-06-26 by niuxjxlnx