Xilinx/Vitis-AI #1166: [Vitis AI 2.0] your GStreamer installation is missing a required plugin

shimoshida, 2023-02-10:
## Settings
- KV260
- Vitis AI 2.0

## Procedure
I prepare for boot image as this README.md 
https://github.com/Xilinx/Vitis-AI/tree/v2.0/setup/mpsoc/VART

and try to display video (*.mp4) by OpenCV, but it does not work.
The following logs are shown.
```
[ WARN:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp (1745) handleMessage OpenCV | GStreamer warning: your GStreamer installation is missing a required plugin
[ WARN:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp (1760) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module uridecodebin0 reported: Your GStreamer installation is missing a plug-in.
[ WARN:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
```

How to solve this problem?

shimoshida, 2023-02-13:
## Vitis AI 2.5で実験
サンプルコード
```
import cv2

#cap = cv2.VideoCapture(0)
cap = cv2.VideoCapture("./aktio_test_mv.mp4")

while True:
    retval,frame = cap.read()
    if not retval:
        cv2.waitKey(10)
        continue
    
    cv2.imshow("yolovx",frame)
    if cv2.waitKey(1)=='q':
        break
```

結果は同じ
```
root@xilinx-kv260-starterkit-20221:~# python3 main.py 
[2023-02-13 07:52:23.749588105]	[omx_core.cpp:181]	[OMX_GetHandle]	Couldnt allocate dma allocator (tried using /dev/allegroDecodeIP)
[ WARN:0] global /usr/src/debug/opencv/4.5.2-r0/git/modules/videoio/src/cap_gstreamer.cpp (2041) handleMessage OpenCV | GStreamer warning: your GStreamer installation is missing a required plugin
[ WARN:0] global /usr/src/debug/opencv/4.5.2-r0/git/modules/videoio/src/cap_gstreamer.cpp (2056) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module qtdemux0 reported: Internal data stream error.
[ WARN:0] global /usr/src/debug/opencv/4.5.2-r0/git/modules/videoio/src/cap_gstreamer.cpp (1034) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /usr/src/debug/opencv/4.5.2-r0/git/modules/videoio/src/cap_gstreamer.cpp (597) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
```

shimoshida, 2023-02-13:
I have tried to experiment this with vitis ai 2.5(`xilinx-kv260-dpu-v2022.1-v2.5.0`), but obtain the same results...
```
root@xilinx-kv260-starterkit-20221:~# python3 main.py 
[2023-02-13 07:52:23.749588105]	[omx_core.cpp:181]	[OMX_GetHandle]	Couldnt allocate dma allocator (tried using /dev/allegroDecodeIP)
[ WARN:0] global /usr/src/debug/opencv/4.5.2-r0/git/modules/videoio/src/cap_gstreamer.cpp (2041) handleMessage OpenCV | GStreamer warning: your GStreamer installation is missing a required plugin
[ WARN:0] global /usr/src/debug/opencv/4.5.2-r0/git/modules/videoio/src/cap_gstreamer.cpp (2056) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module qtdemux0 reported: Internal data stream error.
[ WARN:0] global /usr/src/debug/opencv/4.5.2-r0/git/modules/videoio/src/cap_gstreamer.cpp (1034) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /usr/src/debug/opencv/4.5.2-r0/git/modules/videoio/src/cap_gstreamer.cpp (597) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
```

yangye8, 2023-02-13:
Due to libav's license issue(LGPL), the released sdcard image is unable to parse the mp4 format. There are 2 ways to address it:
1.  Please install libav plugin by yourself
2. convert the video format from .mp4 to .webm. The libvpx is a BSD license and it is installed in the released image.

shimoshida, 2023-02-14:
@yangye8 I have confirmed that `*.webm` works well. Thank you for your help!

closed 2023-02-14 by shimoshida