Xilinx/Vitis-AI #1169: torch.nn.TopK is not support

zyt1024, 2023-02-13:
Hello,The TopK operator is used in the model I want to deploy. I checked the document(https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Operators-Supported-by-PyTorch) and said TopK is not supported. If I want to use Vitis AI,what should I do?

lishixlnx, 2023-02-20:
1. you can use graph runner mode by implementing your own customer op of topk;
please refer to the https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Custom-OP-Workflow part for detail.
2. you can implementation your own logic of topk in post process part.  You can search by "grep -i topk vai_src_tree/ -R " and find lots of topk function in current vai source code.

zyt1024, 2023-02-23:
> 1. you can use graph runner mode by implementing your own customer op of topk;
>    please refer to the https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Custom-OP-Workflow part for detail.
> 2. you can [...]

Thank you, I have read the code about TopK in ` src/vai_library/cpu_task/ops/topk/topk.cpp`, but I found a small error in the provided code during verification. When I  modified this error, the verification was correct. Can I submit a PR to correct this error?

lishixlnx, 2023-03-01:
you can paste your fix here and I'll check it.  thanks.

zyt1024, 2023-03-03:
hello，This is my modified code，I just modified one line
https://github.com/zyt1024/Vitis-AI/commit/4af2ac44f85445f1965ab6630074f865fb038049

zyt1024, 2023-03-03:
> hello，This is my modified code，I just modified one line
> [zyt1024@4af2ac4](https://github.com/zyt1024/Vitis-AI/commit/4af2ac44f85445f1965ab6630074f865fb038049)
hello，This is my modified code，I just modified one line
https://github.com/zyt1024/Vitis-AI/commit/4af2ac44f85445f1965ab6630074f865fb038049

zyt1024, 2023-03-03:
> you can paste your fix here and I'll check it. thanks.

hello，This is my modified code，I just modified one line
https://github.com/zyt1024/Vitis-AI/commit/4af2ac44f85445f1965ab6630074f865fb038049

lishixlnx, 2023-03-07:
your fix is not correct. 
Your fix cause blank gap in the output buffer.

qianglin-xlnx, 2023-06-25:
Closing since no activity for more than 3 months, please reopen if you still have question, thanks

closed 2023-06-25 by qianglin-xlnx