Xilinx/Vitis-AI #1170: replace_pytorch.sh fails - CondaHTTPError: HTTP 000 CONNECTION FAILED

anirudhitagi, 2023-02-13:
Greetings,

I am attempting to install torchaudio and for that I attempted to use the [replace_pytorch.sh](https://github.com/Xilinx/Vitis-AI/blob/v2.5/docker/dockerfiles/replace_pytorch.sh) script but I ran into the error pasted below. (I have cropped parts of the message where there were no error , to save space..... happy to provide the full message if needed)

The main issue with the following packages:
unilog-2.5.0         
target_factory-2.5.0 
vai_utf-2.5.0        
xir-2.5.0            
dpuv3-pycompiler-2.5 
pytorch_nndct_cpu-2. 
vart-2.5.0           
xcompiler-2.5.0      
xnnc-2.5.0           
vaic-2.5.0        

I get a CondaHTTPError: HTTP 000 CONNECTION FAILED for them and I reattempted this process multiple times...

Please advise how to resolve this?

`Vitis-AI /workspace/docker/dockerfiles > ./replace_pytorch.sh vitis-ai-pytorch-torchaudio

#### Creating a new conda environment by cloning vitis-ai-pytorch and activate it...
--2023-02-13 10:01:51--  https://www.xilinx.com/bin/public/openDownload?filename=conda-channel_2.0.0.1103-02.tar.gz
Resolving www.xilinx.com (www.xilinx.com)... 23.199.3.201
Connecting to www.xilinx.com (www.xilinx.com)|23.199.3.201|:443... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: https://xilinx-ax-dl.entitlenow.com/dl/ul/2022/01/17/R210501912/conda-channel_2.0.0.1103-02.tar.gz?hash=tO2sSEi9jg6FevhCc7TeGg&expires=1676340111&filename=conda-channel_2.0.0.1103-02.tar.gz [following]
--2023-02-13 10:01:52--  https://xilinx-ax-dl.entitlenow.com/dl/ul/2022/01/17/R210501912/conda-channel_2.0.0.1103-02.tar.gz?hash=tO2sSEi9jg6FevhCc7TeGg&expires=1676340111&filename=conda-channel_2.0.0.1103-02.tar.gz
Resolving xilinx-ax-dl.entitlenow.com (xilinx-ax-dl.entitlenow.com)... 23.60.84.79
Connecting to xilinx-ax-dl.entitlenow.com (xilinx-ax-dl.entitlenow.com)|23.60.84.79|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 528906896 (504M) [application/x-gzip]
Saving to: ‘conda-channel.tar.gz’

     0K ........ ........ ........ ........ ........ ........  0% 21.8M 23s
  3072K ........ ........ ........ ........ ........ ........  1% 26.6M 21s
  6144K ........ ........ ........ ........ ........ ........  1% 20.8M 22s
  9216K ........ ........ ........ ........ ........ ........  2% 20.0M 22s
 12288K ........ ........ ........ ........ ........ ........  2% 13.2M 25s
 15360K ........ ........ ........ ........ ........ ........  3% 29.7M 24s
 18432K ........ ........ ........ ........ ........ ........  4% 26.9M 23s
 21504K ........ ........ ........ ........ ........ ........  4% 20.8M 23s
 24576K ........ ........ ........ ........ ........ ........  5% 35.4M 21s
 27648K ........ ........ ........ ........ ........ ........  5% 37.4M 20s
 30720K ........ ........ ........ ........ ........ ........  6% 26.1M 20s
 33792K ........ ........ ........ ........ ........ ........  7% 23.1M 20s
 36864K ........ ........ ........ ........ ........ ........  7% 30.9M 20s
 39936K ........ ........ ........ ........ ........ ........  8% 27.2M 19s
 43008K ........ ........ ........ ........ ........ ........  8% 35.3M 19s
 46080K ........ ........ ........ ........ ........ ........  9% 25.0M 19s
 49152K ........ ........ ........ ........ ........ ........ 10% 17.0M 19s
 52224K ........ ........ ........ ........ ........ ........ 10% 43.1M 18s
 55296K ........ ........ ........ ........ ........ ........ 11% 31.4M 18s
 58368K ........ ........ ........ ........ ........ ........ 11% 25.9M 18s
 61440K ........ ........ ........ ........ ........ ........ 12% 29.5M 18s
 64512K ........ ........ ........ ........ ........ ........ 13% 24.2M 17s
 67584K ........ ........ ........ ........ ........ ........ 13% 27.0M 17s
 70656K ........ ........ ........ ........ ........ ........ 14% 20.6M 17s
 73728K ........ ........ ........ ........ ........ ........ 14% 20.4M 17s
 76800K ........ ........ ........ ........ ........ ........ 15% 36.5M 17s
 79872K ........ ........ ........ ........ ........ ........ 16% 19.5M 17s
 82944K ........ ........ ........ ........ ........ ........ 16% 25.2M 17s
 86016K ........ ........ ........ ........ ........ ........ 17% 31.8M 17s
 89088K ........ ........ ........ ........ ........ ........ 17% 13.8M 17s
 92160K ........ ........ ........ ........ ........ ........ 18% 12.9M 17s
 95232K ........ ........ ........ ........ ........ ........ 19% 35.3M 17s
 98304K ........ ........ ........ ........ ........ ........ 19% 31.0M 17s
101376K ........ ........ ........ ........ ........ ........ 20% 35.2M 17s
104448K ........ ........ ........ ........ ........ ........ 20% 41.1M 16s
[... 126 lines cut ...]
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <http://artifactory/artifactory/vitis-ai-conda-release/linux-64/xcompiler-2.5.0-py37hea4fdf2_144.tar.bz2>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <http://artifactory/artifactory/vitis-ai-conda-release/linux-64/xnnc-2.5.0-py37hd51751d_116.tar.bz2>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <http://artifactory/artifactory/vitis-ai-conda-release/linux-64/vaic-2.5.0-py37hd51751d_48.tar.bz2>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.

#### New conda environment is NOT created correctly.
`

niuxjxlnx, 2023-02-14:
@anirudhitagi ,

There is a fix for the script replace_pytorch.sh after v2.5. 
In vitis-ai 2.5 docker, you can try this fix at https://github.com/Xilinx/Vitis-AI/blob/06a72491bd9fc72bbc743ea1753d005cbeb6acc8/docker/dockerfiles/replace_pytorch.sh.
Also you can try vitis-ai 3.0 docker, in which the script link is https://github.com/Xilinx/Vitis-AI/blob/v3.0/docker/common/replace_pytorch.sh

anirudhitagi, 2023-02-14:
Hi @niuxjxlnx,
Thank you for your response I tried the solution you suggested for:

> In vitis-ai 2.5 docker, you can try this fix at https://github.com/Xilinx/Vitis-AI/blob/06a72491bd9fc72bbc743ea1753d005cbeb6acc8/docker/dockerfiles/replace_pytorch.sh.

I got the following error:

[replace_pytorch-new_error.log](https://github.com/Xilinx/Vitis-AI/files/10735964/replace_pytorch-new_error.log)

Here is the summary of the error:

```
Vitis-AI /workspace > docker/dockerfiles/replace_pytorch.sh vitis-ai-pytorch-replaced-neww

#### Creating a new conda environment by cloning vitis-ai-pytorch and activate it...
--2023-02-14 11:47:16--  https://www.xilinx.com/bin/public/openDownload?filename=conda-channel_2.5.0.1260-01.tar.gz
Resolving www.xilinx.com (www.xilinx.com)... 104.92.8.33
Connecting to www.xilinx.com (www.xilinx.com)|104.92.8.33|:443... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: https://xilinx-ax-dl.entitlenow.com/dl/ul/2022/06/13/R210599437/conda-channel_2.5.0.1260-01.tar.gz?hash=PC7GBJ_rkWNgfm2-72hSNQ&expires=1676432836&filename=conda-channel_2.5.0.1260-01.tar.gz [following]
--2023-02-14 11:47:16--  https://xilinx-ax-dl.entitlenow.com/dl/ul/2022/06/13/R210599437/conda-channel_2.5.0.1260-01.tar.gz?hash=PC7GBJ_rkWNgfm2-72hSNQ&expires=1676432836&filename=conda-channel_2.5.0.1260-01.tar.gz
Resolving xilinx-ax-dl.entitlenow.com (xilinx-ax-dl.entitlenow.com)... 23.201.212.73
Connecting to xilinx-ax-dl.entitlenow.com (xilinx-ax-dl.entitlenow.com)|23.201.212.73|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 498369241 (475M) [application/x-gzip]
Saving to: ‘conda-channel.tar.gz’

     0K ........ ........ ........ ........ ........ ........  0% 11.1M 43s
  3072K ........ ........ ........ ........ ........ ........  1% 11.2M 42s
  6144K ........ ........ ........ ........ ........ ........  1% 11.2M 42s
................................................................................................
476160K ........ ........ ........ ........ ........ ........ 98% 11.2M 1s
479232K ........ ........ ........ ........ ........ ........ 99% 11.2M 0s
482304K ........ ........ ........ ........ ........ ........ 99% 11.0M 0s
485376K ........ ........ ....                               100% 11.3M=43s

2023-02-14 11:48:00 (11.1 MB/s) - ‘conda-channel.tar.gz’ saved [498369241/498369241]

conda-channel/
conda-channel/linux-64/
conda-channel/linux-64/vai_optimizer_darknet_gpu-2.5.0-h8661612_31.tar.bz2
conda-channel/linux-64/dpuv3-pycompiler-2.5.0-py37h3f44cd5_108.tar.bz2
conda-channel/linux-64/dpuv3-pycompiler-2.5.0-py36h3f44cd5_108.tar.bz2
conda-channel/linux-64/tf_nndct-2.5.0-py36h5abb5ad_41.tar.bz2
conda-channel/linux-64/pytorch_nndct_lstm-2.5.0-py36h67eafb0_41.tar.bz2
conda-channel/linux-64/pytorch_nndct_cpu-2.5.0-py37h05a16ff_108.tar.bz2
conda-channel/linux-64/pytorch_nndct-2.5.0-py37h4d48dc7_108.tar.bz2
.....................................................................................................

Looking in links: https://download.pytorch.org/whl/torch_stable.html
Collecting torchvision==0.8.2+cpu
  Downloading https://download.pytorch.org/whl/cpu/torchvision-0.8.2%2Bcpu-cp37-cp37m-linux_x86_64.whl (11.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 11.8/11.8 MB 11.2 MB/s eta 0:00:00
Requirement already satisfied: torch==1.7.1 in /opt/vitis_ai/conda/envs/vitis-ai-pytorch-replaced-neww/lib/python3.7/site-packages (from torchvision==0.8.2+cpu) (1.7.1)
Requirement already satisfied: pillow>=4.1.1 in /opt/vitis_ai/conda/envs/vitis-ai-pytorch-replaced-neww/lib/python3.7/site-packages (from torchvision==0.8.2+cpu) (8.1.0)
Requirement already satisfied: numpy in /opt/vitis_ai/conda/envs/vitis-ai-pytorch-replaced-neww/lib/python3.7/site-packages (from torchvision==0.8.2+cpu) (1.17.5)
Requirement already satisfied: typing_extensions in /opt/vitis_ai/conda/envs/vitis-ai-pytorch-replaced-neww/lib/python3.7/site-packages (from torch==1.7.1->torchvision==0.8.2+cpu) (4.2.0)
Installing collected packages: torchvision
Successfully installed torchvision-0.8.2+cpu

#### Pytorch packages is replaced successfully.

#### Checkout code of vai_q_pytorch ...
>>>> You can apply your local code of vai_q_pytorch and comment out the following lines of git command <<<<
Initialized empty Git repository in /scratch/code_vaiq/.git/
remote: Enumerating objects: 74230, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 74230 (delta 0), reused 0 (delta 0), pack-reused 74229
Receiving objects: 100% (74230/74230), 2.04 GiB | 11.16 MiB/s, done.
Resolving deltas: 100% (32687/32687), done.
From https://github.com/Xilinx/Vitis-AI
 * branch                master     -> FETCH_HEAD
 * [new branch]          master     -> origin/master
error: Sparse checkout leaves no entry on working directory
docker/dockerfiles/replace_pytorch.sh: line 110: cd: src/Vitis-AI-Quantizer/vai_q_pytorch/: No such file or directory

#### Vai_q_pytorch code is NOT checked out successfully.

```

Please advise how to proceed?

anirudhitagi, 2023-02-14:
Also I noticed torchaudio is not mentioned in the replace_pytorch.sh script? How may I add that?

niuxjxlnx, 2023-02-16:
@anirudhitagi ,

The code path changed on github.
You can replace the following string in replace_pytorch.sh from:
`src/Vitis-AI-Quantizer/vai_q_pytorch`
to
`src/vai_quantizer/vai_q_pytorch`

Another approach to avoid the version mismatch problem is to try vitis-ai 3.0 directly. In vitis-ai 3.0, the script is https://github.com/Xilinx/Vitis-AI/blob/master/docker/common/replace_pytorch.sh

niuxjxlnx, 2023-02-16:
> Also I noticed torchaudio is not mentioned in the replace_pytorch.sh script? How may I add that?

You can add it into the script (edit the line to install torchvision) according to the instructions in this page https://pytorch.org/get-started/previous-versions/.

anirudhitagi, 2023-02-16:
Hi @niuxjxlnx Thank you so much for the suggestions. This has worked!

I made the changes to the `replace_pytorch.sh` as you recommended and added the following version to torch installation.

```
  mamba install -y pytorch==1.12.1 cpuonly -c pytorch
  pip install torchvision==0.13.1+cpu -f https://download.pytorch.org/whl/torch_stable.html
  pip install torchaudio==0.12.1+cpu -f https://download.pytorch.org/whl/torch_stable.html

```

It appears to have installed correctly :

```
(vitis-ai-pytorch-neww) Vitis-AI /workspace > python3
Python 3.7.12 | packaged by conda-forge | (default, Oct 26 2021, 06:08:53) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> import torchaudio
>>> import torchvision
>>> from pytorch_nndct.apis import torch_quantizer
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'

[VAIQ_NOTE]: Loading NNDCT kernels...

```

closed 2023-02-16 by anirudhitagi