Xilinx/Vitis-AI #1171: docker_build_gpu.sh failed on v2.5

anirudhitagi, 2023-02-13:
Greetings,

I am using Vitis-AI v2.5 on Ubuntu 20.04
Here is the output of my nvidia-smi

`Mon Feb 13 17:14:12 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.60.13    Driver Version: 525.60.13    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   40C    P0    N/A /  N/A |      6MiB /  6144MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1113      G   /usr/lib/xorg/Xorg                  4MiB |
+-----------------------------------------------------------------------------+`

and I ran the following steps :

```
git clone --recursive https://github.com/Xilinx/Vitis-AI.git -b v2.5
cd Vitis-AI/docker
./docker_build_gpu.sh
```
But I got the following error:

`Do you agree to the terms and wish to proceed [y/n]? y
[+] Building 0.4s (3/3) FINISHED                                                                                                                                                                            
 => [internal] load build definition from vitis-ai-gpu.Dockerfile                                                                                                                                      0.0s
 => => transferring dockerfile: 14.13kB                                                                                                                                                                0.0s
 => [internal] load .dockerignore                                                                                                                                                                      0.0s
 => => transferring context: 2B                                                                                                                                                                        0.0s
 => ERROR [internal] load metadata for docker.io/nvidia/cuda:11.0-cudnn8-runtime-ubuntu18.04                                                                                                           0.4s
------
 > [internal] load metadata for docker.io/nvidia/cuda:11.0-cudnn8-runtime-ubuntu18.04:
------
vitis-ai-gpu.Dockerfile:1
--------------------
   1 | >>> FROM nvidia/cuda:11.0-cudnn8-runtime-ubuntu18.04
   2 |     env DEBIAN_FRONTEND=noninteractive
   3 |     SHELL ["/bin/bash", "-c"]
--------------------
ERROR: failed to solve: nvidia/cuda:11.0-cudnn8-runtime-ubuntu18.04: docker.io/nvidia/cuda:11.0-cudnn8-runtime-ubuntu18.04: not found
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103`

Please advise how to resolve this?

janifer112x, 2023-02-15:
Hi @anirudhitagi ,
Could you try to checkout 2.5 branch instead of v2.5 tag, we have all fixes after 2.5 released in that branch.

git checkout 2.5

anirudhitagi, 2023-02-16:
Hi @janifer112x ,
Thank you for the suggestion. While your suggestion is progressing a little more than earlier, for some reason my PC is crashing midway through this process.... I am still trying to debug this. I will update here once I do.
Thank you.