Xilinx/Vitis-AI #1172: Vitis-AI2.5 ,XCZU5EV,  Run AI Applications ， [XRT]ERROR:DPUCZDX8G:DPUCZDX8G 1 not found

LoeWHJ, 2023-02-14:
Hi
When I use Vitis-AI2.5 to execute Vitis-AI-Tutorials/Design_Tutorials/09-mnist_pyt on the core board of XCZU5EV model, I encounter [XRT]ERROR:DPUCZDX8G:DPUCZDX8G_1 not found problem.
The process is as follows:
1 Use Vitis-AI1.4 training model
2Use Vitis-AI2.5 to quantify and compile the model
3 The arch.json used in compiling the model is compiled using vitis2022.1 and DPUCZDX8G 4.0
4 The following error occurs when the compiled model is deployed to the board for execution

root@xczu5evpetalinux:/run/media/mmcblk1p1/target_xczu5ev#python3 app_mt.py -m CNN_xczu5ev.xmodel
Command line options:
--image dir images
--threads
--model
CNN xczu5ev.xmodel
XRT build version:2.13.0
Build hash:2a6dc026480914ea1c9f02977a6ab4b57e8a3c8d
Build date:2022-04-1214:31:55
Git branch:2022.1
PID:782
UID:0
[Fri Nov1917:41:592021GMT]
HOST:
EXE:/usr/bin/python3.9
[XRT]ERROR:DPUCZDX8G:DPUCZDX8G 1 not found
[XRT]ERROR:xclRegRW:invalid CU index:-2
WARNING:Logging before InitGoogleLogging()is written to STDERR
F111917:41:59.327543
782 xrt_device_handle_imp.cpp:96]Check failed:read result ==0 (-22 vs.0)xclRead has error!
**Check failure stack trace:**
Aborted

qianglin-xlnx, 2023-02-16:
@LoeWHJ 
You can run "xdputil query" to check the dpu.
If the dpu core can't be detected, please check the /etc/vart.conf file to make sure the dpu bit file path is set correctly.

MohamedRaizz, 2023-02-21:
@LoeWHJ @qianglin-xlnx  
                         I try to run AI application I also got same issue and i check /etc/vart.conf file that dpu path is correctly i hope u can help me....

root@Petalinux:/run/media/sda1/facedetect# ./test_video_facedetect cf_densebox_wider_320_320_0.49G_2.0 test_45.mp4 -t 1
XRT build version: 2.14.0
Build hash: 43926231f7183688add2dccfd391b36a1f000bea[ 1358.108242] audit: type=1701 audit(1637343717.748:51): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=1544 comm="test_video_face" exe="/run/media/sd1

Build date: 2022-10-07 05:12:02
Git branch: 2022.2
PID: 1544
UID: 0
[Fri Nov 19 17:41:57 2021 GMT]
HOST: 
EXE: /run/media/sda1/facedetect21.1/test_video_facedetect
[XRT] ERROR: DPUCZDX8G:DPUCZDX8G_1 not found
[XRT] ERROR: xclRegRW: invalid CU index: -2
WARNING: Logging before InitGoogleLogging() is written to STDERR
F1119 17:41:57.751024  1544 xrt_device_handle_imp.cpp:96] Check failed: read_result == 0 (-22 vs. 0) xclRead has error!
*** Check failure stack trace: ***
Aborted
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
root@Petalinux:/run/media/sda1/facedetect# xdputil query
XRT build version: 2.14.0
Build hash: 43926231f7183688add2dccfd391b36a1f000bea[ 1430.106976] audit: type=1701 audit(1637343789.748:52): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=1551 comm="python3" exe="/usr/bin/python3.9" s1

Build date: 2022-10-07 05:12:02
Git branch: 2022.2
PID: 1551
UID: 0
[Fri Nov 19 17:43:09 2021 GMT]
HOST: 
EXE: /usr/bin/python3.9
[XRT] ERROR: DPUCZDX8G:DPUCZDX8G_1 not found
[XRT] ERROR: xclRegRW: invalid CU index: -2
WARNING: Logging before InitGoogleLogging() is written to STDERR
F1119 17:43:09.749843  1551 xrt_device_handle_imp.cpp:96] Check failed: read_result == 0 (-22 vs. 0) xclRead has error!
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:  1551 Aborted                 /usr/bin/python3 -m xdputil $*
root@Petalinux:/run/media/sda1/facedetect#

LoeWHJ, 2023-02-27:
@MohamedRaizz 
Have you solved the problem and can you describe in detail how you built the system with petalinux and what configuration changes you made, such as in the file under recipes-kernel？

MohamedRaizz, 2023-03-01:
@LoeWHJ @qianglin-xlnx I did not solve this problem i refer this below link https://support.xilinx.com/s/question/0D52E00007H41eaSAB/how-to-report-bug-in-vitis-ai-25-for-mpsoc-and-send-a-patch-file-?language=en_US 
here they told to add dpu patch in recipies-kernel it is also not working.

qianglin-xlnx, 2023-06-25:
Hi @MohamedRaizz 
Sorry for the late reply. Is your issue resolved?
Usually, for this issue, we need to check if zocl driver is installed correctly. You can also check the system.dtb device tree if the interrupts are set correctly.

MohamedRaizz, 2023-07-07:
> Hi @MohamedRaizz Sorry for the late reply. Is your issue resolved? Usually, for this issue, we need to check if zocl driver is installed correctly. You can also check the system.dtb device tree if the interrupts are set correctly.

Hi @LoeWHJ @qianglin-xlnx 
No is still issue was not solved 
I refer the below link for enable zocl driver in rootfs configuration 
https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/Vitis_Platform_Creation/Feature_Tutorials/02_petalinux_customization/README.md

can u please tell me what configurations  we need to check

MohamedRaizz, 2023-07-07:
> Hi @MohamedRaizz Sorry for the late reply. Is your issue resolved? Usually, for this issue, we need to check if zocl driver is installed correctly. You can also check the system.dtb device tree if the interrupts are set correctly.

@LoeWHJ @qianglin-xlnx 

below i shared inside components path **pl.dtsi**  for zocl node
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
                dpu_intc: interrupt-controller@80100000 {
                        #interrupt-cells = <2>;
                        clock-names = "s_axi_aclk";
                        clocks = <&zynqmp_clk 71>;
                        compatible = "xlnx,axi-intc-4.1", "xlnx,xps-intc-1.00.a";
                        interrupt-controller ;
                        reg = <0x0 0x80100000 0x0 0x10000>;
                        xlnx,kind-of-intr = <0x1>;
                        xlnx,num-intr-inputs = <0x20>;
                };
                zyxclmm_drm {
                        compatible = "xlnx,zocl";
                        interrupts-extended = <&dpu_intc 0 4>, <&dpu_intc 1 4>, <&dpu_intc 2 4>, <&dpu_intc 3 4>, <&dpu_intc 4 4>, <&dpu_intc 5 4>, <&dpu_intc 6 4>, <&dpu_intc 7 4>, <&dpu_intc 8 4>, <&dpu_intc 9 4>,
<&dpu_intc 10 4>, <&dpu_intc 11 4>, <&dpu_intc 12 4>, <&dpu_intc 13 4>, <&dpu_intc 14 4>,
<&dpu_intc 15 4>, <&dpu_intc 16 4>, <&dpu_intc 17 4>, <&dpu_intc 18 4>, <&dpu_intc 19 4>,
<&dpu_intc 20 4>, <&dpu_intc 21 4>, <&dpu_intc 22 4>, <&dpu_intc 23 4>, <&dpu_intc 24 4>,
<&dpu_intc 25 4>, <&dpu_intc 26 4>, <&dpu_intc 27 4>, <&dpu_intc 28 4>, <&dpu_intc 29 4>,
<&dpu_intc 30 4>, <&dpu_intc 31 4 >;
                };
        };
};
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
**system-user.dtsi** 

        dpu_intc: interrupt-controller@80100000 {
                #interrupt-cells = <2>;
                compatible = "xlnx,axi-intc-4.1", "xlnx,xps-intc-1.00.a";
                interrupt-controller ;
                reg = <0x0 0x80100000 0x0 0x10000>;
                xlnx,kind-of-intr = <0x1>;
                xlnx,num-intr-inputs = <0x20>;
                interrupt-parent = <&gic>;
                interrupts = <0 90 4>;
        };

        zyxclmm_drm {
                compatible = "xlnx,zocl";
                status = "okay";
                interrupt-parent = <&dpu_intc>;
                interrupts = <0 4>, <1 4>, <2 4>, <3 4>,
                     <4 4>, <5 4>, <6 4>, <7 4>,
                     <8 4>, <9 4>, <10 4>, <11 4>,
                     <12 4>, <13 4>, <14 4>, <15 4>,
                     <16 4>, <17 4>, <18 4>, <19 4>,
                     <20 4>, <21 4>, <22 4>, <23 4>,
                     <24 4>, <25 4>, <26 4>, <27 4>,
                     <28 4>, <29 4>, <30 4>, <31 4>;
        };
};
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
After  petalinux compilation inside images/linux **system.dtb** that i convert to dts

                interrupt-controller@80100000 {
                        #interrupt-cells = <0x02>;
                        clock-names = "s_axi_aclk";
                        clocks = <0x03 0x47>;
                        compatible = "xlnx,axi-intc-4.1\0xlnx,xps-intc-1.00.a";
                        interrupt-controller;
                        reg = <0x00 0x80100000 0x00 0x10000>;
                        xlnx,kind-of-intr = <0x01>;
                        xlnx,num-intr-inputs = <0x20>;
                        interrupt-parent = <0x04>;
                        interrupts = <0x00 0x5a 0x04>;
                        phandle = <0x20>;
                };

                zyxclmm_drm {
                        compatible = "xlnx,zocl";
                        interrupts-extended = <0x20 0x00 0x04 0x20 0x01 0x04 0x20 0x02 0x04 0x20 0x03 0x04 0x20 0x04 0x04 0x20 0x05 0x04 0x20 0x06 0x04 0x20 0x07 0x04 0x20 0x08 0x04 0x20 0x09 0x04 0x20 0x0a 0x04 0x20 0x0b 0x04 0x20 0x0c 0x04 0x20 0x0d 0x04 0x20 0x0e 0x04 0x20 0x0f 0x04 0x20 0x10 0x04 0x20 0x11 0x04 0x20 0x12 0x04 0x20 0x13 0x04 0x20 0x14 0x04 0x20 0x15 0x04 0x20 0x16 0x04 0x20 0x17 0x04 0x20 0x18 0x04 0x20 0x19 0x04 0x20 0x1a 0x04 0x20 0x1b 0x04 0x20 0x1c 0x04 0x20 0x1d 0x04 0x20 0x1e 0x04 0x20 0x1f 0x04>;
                        status = "okay";
                        interrupt-parent = <0x20>;
                        interrupts = <0x00 0x04 0x01 0x04 0x02 0x04 0x03 0x04 0x04 0x04 0x05 0x04 0x06 0x04 0x07 0x04 0x08 0x04 0x09 0x04 0x0a 0x04 0x0b 0x04 0x0c 0x04 0x0d 0x04 0x0e 0x04 0x0f 0x04 0x10 0x04 0x11 0x04 0x12 0x04 0x13 0x04 0x14 0x04 0x15 0x04 0x16 0x04 0x17 0x04 0x18 0x04 0x19 0x04 0x1a 0x04 0x1b 0x04 0x1c 0x04 0x1d 0x04 0x1e 0x04 0x1f 0x04>;
                };

                som_rev {
                        som-rev-gpios = <0x10 0x4e 0x01 0x10 0x4f 0x01 0x10 0x50 0x01 0x10 0x51 0x01 0x10 0x52 0x01 0x10 0x53 0x01 0x10 0x54 0x01 0x10 0x55 0x01>;
                };
        };
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
can u please check that inerrupts and provide me an update

MohamedRaizz, 2023-07-07:
> Hi @MohamedRaizz Sorry for the late reply. Is your issue resolved? Usually, for this issue, we need to check if zocl driver is installed correctly. You can also check the system.dtb device tree if the interrupts are set correctly.

below i shared inside components path pl.dtsi for zocl node
            dpu_intc: interrupt-controller@80100000 {
                    #interrupt-cells = <2>;
                    clock-names = "s_axi_aclk";
                    clocks = <&zynqmp_clk 71>;
                    compatible = "xlnx,axi-intc-4.1", "xlnx,xps-intc-1.00.a";
                    interrupt-controller ;
                    reg = <0x0 0x80100000 0x0 0x10000>;
                    xlnx,kind-of-intr = <0x1>;
                    xlnx,num-intr-inputs = <0x20>;
            };
            zyxclmm_drm {
                    compatible = "xlnx,zocl";
                    interrupts-extended = <&dpu_intc 0 4>, <&dpu_intc 1 4>, <&dpu_intc 2 4>, <&dpu_intc 3 4>, <&dpu_intc 4 4>, <&dpu_intc 5 4>, <&dpu_intc 6 4>, <&dpu_intc 7 4>, <&dpu_intc 8 4>, <&dpu_intc 9 4>,
<&dpu_intc 10 4>, <&dpu_intc 11 4>, <&dpu_intc 12 4>, <&dpu_intc 13 4>, <&dpu_intc 14 4>,
<&dpu_intc 15 4>, <&dpu_intc 16 4>, <&dpu_intc 17 4>, <&dpu_intc 18 4>, <&dpu_intc 19 4>,
<&dpu_intc 20 4>, <&dpu_intc 21 4>, <&dpu_intc 22 4>, <&dpu_intc 23 4>, <&dpu_intc 24 4>,
<&dpu_intc 25 4>, <&dpu_intc 26 4>, <&dpu_intc 27 4>, <&dpu_intc 28 4>, <&dpu_intc 29 4>,
<&dpu_intc 30 4>, <&dpu_intc 31 4 >;
};
};
};
system-user.dtsi

    dpu_intc: interrupt-controller@80100000 {
            #interrupt-cells = <2>;
            compatible = "xlnx,axi-intc-4.1", "xlnx,xps-intc-1.00.a";
            interrupt-controller ;
            reg = <0x0 0x80100000 0x0 0x10000>;
            xlnx,kind-of-intr = <0x1>;
            xlnx,num-intr-inputs = <0x20>;
            interrupt-parent = <&gic>;
            interrupts = <0 90 4>;
    };

    zyxclmm_drm {
            compatible = "xlnx,zocl";
            status = "okay";
            interrupt-parent = <&dpu_intc>;
            interrupts = <0 4>, <1 4>, <2 4>, <3 4>,
                 <4 4>, <5 4>, <6 4>, <7 4>,
                 <8 4>, <9 4>, <10 4>, <11 4>,
                 <12 4>, <13 4>, <14 4>, <15 4>,
                 <16 4>, <17 4>, <18 4>, <19 4>,
                 <20 4>, <21 4>, <22 4>, <23 4>,
                 <24 4>, <25 4>, <26 4>, <27 4>,
                 <28 4>, <29 4>, <30 4>, <31 4>;
    };
};
After petalinux compilation inside images/linux system.dtb that i convert to dts

            interrupt-controller@80100000 {
                    #interrupt-cells = <0x02>;
                    clock-names = "s_axi_aclk";
                    clocks = <0x03 0x47>;
                    compatible = "xlnx,axi-intc-4.1\0xlnx,xps-intc-1.00.a";
                    interrupt-controller;
                    reg = <0x00 0x80100000 0x00 0x10000>;
                    xlnx,kind-of-intr = <0x01>;
                    xlnx,num-intr-inputs = <0x20>;
                    interrupt-parent = <0x04>;
                    interrupts = <0x00 0x5a 0x04>;
                    phandle = <0x20>;
            };

            zyxclmm_drm {
                    compatible = "xlnx,zocl";
                    interrupts-extended = <0x20 0x00 0x04 0x20 0x01 0x04 0x20 0x02 0x04 0x20 0x03 0x04 0x20 0x04 0x04 0x20 0x05 0x04 0x20 0x06 0x04 0x20 0x07 0x04 0x20 0x08 0x04 0x20 0x09 0x04 0x20 0x0a 0x04 0x20 0x0b 0x04 0x20 0x0c 0x04 0x20 0x0d 0x04 0x20 0x0e 0x04 0x20 0x0f 0x04 0x20 0x10 0x04 0x20 0x11 0x04 0x20 0x12 0x04 0x20 0x13 0x04 0x20 0x14 0x04 0x20 0x15 0x04 0x20 0x16 0x04 0x20 0x17 0x04 0x20 0x18 0x04 0x20 0x19 0x04 0x20 0x1a 0x04 0x20 0x1b 0x04 0x20 0x1c 0x04 0x20 0x1d 0x04 0x20 0x1e 0x04 0x20 0x1f 0x04>;
                    status = "okay";
                    interrupt-parent = <0x20>;
                    interrupts = <0x00 0x04 0x01 0x04 0x02 0x04 0x03 0x04 0x04 0x04 0x05 0x04 0x06 0x04 0x07 0x04 0x08 0x04 0x09 0x04 0x0a 0x04 0x0b 0x04 0x0c 0x04 0x0d 0x04 0x0e 0x04 0x0f 0x04 0x10 0x04 0x11 0x04 0x12 0x04 0x13 0x04 0x14 0x04 0x15 0x04 0x16 0x04 0x17 0x04 0x18 0x04 0x19 0x04 0x1a 0x04 0x1b 0x04 0x1c 0x04 0x1d 0x04 0x1e 0x04 0x1f 0x04>;
            };

            som_rev {
                    som-rev-gpios = <0x10 0x4e 0x01 0x10 0x4f 0x01 0x10 0x50 0x01 0x10 0x51 0x01 0x10 0x52 0x01 0x10 0x53 0x01 0x10 0x54 0x01 0x10 0x55 0x01>;
            };
    };

andrew0917, 2023-07-31:
Hi @MohamedRaizz , 
As for the system.dtb, it looks that the zocl parent interrupt controller configuration duplicated, so you can  just use once enough, In addition, there is no interrupt-parent and interrupts propery for  dpu_intc: interrupt-controller@80100000 in the pl.dtsi, do you check the  interrupt-controller have connected the PS gic controller in vivado block design.

qianglin-xlnx, 2023-09-01:
Closing since no activity for more than 1 month, please reopen if you still have question, thanks

closed 2023-09-01 by qianglin-xlnx