Xilinx/Vitis-AI #1175: torch.int64 and torch.int32 not quantizable?

anirudhitagi, 2023-02-16:
Greetings,

I am attempting to quantize the torchvision.models.vit_b_16 model.
I am working on Vitis-AI 2.5, with torch==1.12.1 and torchvision==0.13.1+cpu.
When I attempt to quantize this model I get the following error - 
[vit_b_16_quantization_error.txt](https://github.com/Xilinx/Vitis-AI/files/10759735/vit_b_16_quantization_error.txt)

I see the following warnings - 
```
[VAIQ_WARN][QUANTIZER_TORCH_FLOAT_OP]: The quantizer recognize new op `nndct_expand` as a float operator by default.

[VAIQ_WARN][QUANTIZER_TORCH_FLOAT_OP]: The quantizer recognize new op `nndct_bmm` as a float operator by default.

[VAIQ_WARN][QUANTIZER_TORCH_FLOAT_OP]: The quantizer recognize new op `nndct_select` as a float operator by default.

[VAIQ_WARN][QUANTIZER_TORCH_FLOAT_OP]: The quantizer recognize new op `aten::div` as a float operator by default.

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/VisionTransformer.py)

[VAIQ_NOTE]: =>Get module with quantization.
  0%|                                                                                                                                                                               | 0/100 [00:00<?, ?it/s]
[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_TYPE_NOT_QUANTIZABLE]: The tensor type of VisionTransformer::VisionTransformer/12391 is torch.int64. Only support float32/double/float16 quantization.

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_TYPE_NOT_QUANTIZABLE]: The tensor type of VisionTransformer::VisionTransformer/Encoder[encoder]/Sequential[layers]/EncoderBlock[encoder_layer_0]/MultiheadAttention[self_attention]/14809 is torch.int64. Only support float32/double/float16 quantization.

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_TYPE_NOT_QUANTIZABLE]: The tensor type of VisionTransformer::VisionTransformer/Encoder[encoder]/Sequential[layers]/EncoderBlock[encoder_layer_0]/MultiheadAttention[self_attention]/14811 is torch.int64. Only support float32/double/float16 quantization.

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_TYPE_NOT_QUANTIZABLE]: The tensor type of VisionTransformer::VisionTransformer/Encoder[encoder]/Sequential[layers]/EncoderBlock[encoder_layer_0]/MultiheadAttention[self_attention]/14813 is torch.int64. Only support float32/double/float16 quantization.

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_TYPE_NOT_QUANTIZABLE]: The tensor type of VisionTransformer::VisionTransformer/Encoder[encoder]/Sequential[layers]/EncoderBlock[encoder_layer_0]/MultiheadAttention[self_attention]/12591 is torch.int32. Only support float32/double/float16 quantization.
....................................................................................................................................................................................................................................................................................................................................................

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_TYPE_NOT_QUANTIZABLE]: The tensor type of VisionTransformer::VisionTransformer/Encoder[encoder]/Sequential[layers]/EncoderBlock[encoder_layer_11]/MultiheadAttention[self_attention]/14725 is torch.int32. Only support float32/double/float16 quantization.
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 100/100 [04:41<00:00,  2.82s/it]
loss: 3.23606
top-1 / top-5 accuracy: 51 / 70

[VAIQ_NOTE]: =>Exporting quant config.(quantize_result/quant_info.json)

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_NOT_QUANTIZED]: Node ouptut tensor is not quantized: VisionTransformer::VisionTransformer/12391 type: nndct_elemwise_mul

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_NOT_QUANTIZED]: Node ouptut tensor is not quantized: VisionTransformer::VisionTransformer/Encoder[encoder]/Sequential[layers]/EncoderBlock[encoder_layer_0]/MultiheadAttention[self_attention]/14809 type: nndct_elemwise_mul

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_NOT_QUANTIZED]: Node ouptut tensor is not quantized: VisionTransformer::VisionTransformer/Encoder[encoder]/Sequential[layers]/EncoderBlock[encoder_layer_0]/MultiheadAttention[self_attention]/14811 type: nndct_elemwise_mul

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_NOT_QUANTIZED]: Node ouptut tensor is not quantized: VisionTransformer::VisionTransformer/Encoder[encoder]/Sequential[layers]/EncoderBlock[encoder_layer_0]/MultiheadAttention[self_attention]/14813 type: nndct_elemwise_mul
....................................................................................................................................................................................................................................................................................................................................................

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_NOT_QUANTIZED]: Node ouptut tensor is not quantized: VisionTransformer::VisionTransformer/Encoder[encoder]/Sequential[layers]/EncoderBlock[encoder_layer_11]/MultiheadAttention[self_attention]/14725 type: nndct_elemwise_mul

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_NOT_QUANTIZED]: Parameter tensor is not quantized: VisionTransformer::encoder.pos_embedding. If this parameter is not a parameter embedded in torch operation, like weights of CONV, this kind of parameter will not be quantized. Please embed the parameter into CONV/Linear/BN.

[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_NOT_QUANTIZED]: Some tensors are not quantized, please check their particularity.

....................................................................................................................................................................................................................................................................................................................................................

/opt/vitis_ai/conda/envs/vitis-ai-pytorch-neww/lib/python3.7/site-packages/pytorch_nndct/nn/modules/module_template.py:132: TracerWarning: torch.tensor results are registered as constants in the trace. You can safely ignore this warning if you use this function to create tensors out of constant variables that would be the same every time you call this function. In any other case, this might cause the trace to be incorrect.
  output = caller(*args, **kwargs)
/opt/vitis_ai/conda/envs/vitis-ai-pytorch-neww/lib/python3.7/site-packages/pytorch_nndct/quantization/torchquantizer.py:53: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  if inf.sum() > 0 or nan.sum() > 0:
/opt/vitis_ai/conda/envs/vitis-ai-pytorch-neww/lib/python3.7/site-packages/pytorch_nndct/nn/modules/fix_ops.py:68: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  tensor.storage().size() != tensor.numel()):
/opt/vitis_ai/conda/envs/vitis-ai-pytorch-neww/lib/python3.7/site-packages/pytorch_nndct/nn/modules/prim_ops.py:116: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  if not (list(self.node.out_tensors[0].shape[1:]) == list(input.size())[1:]):
/opt/vitis_ai/conda/envs/vitis-ai-pytorch-neww/lib/python3.7/site-packages/pytorch_nndct/nn/modules/module_template.py:132: UserWarning: To copy construct from a tensor, it is recommended to use sourceTensor.clone().detach() or sourceTensor.clone().detach().requires_grad_(True), rather than torch.tensor(sourceTensor).
  output = caller(*args, **kwargs)
/opt/vitis_ai/conda/envs/vitis-ai-pytorch-neww/lib/python3.7/site-packages/pytorch_nndct/nn/modules/prim_ops.py:51: TracerWarning: Converting a tensor to a Python integer might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  output = int(input)
[... 19 lines cut ...]
[... cut ...]
-------- End of vit_b_16 test 
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][FATAL][XCOM_FILE_NOT_EXISTS][The file is not exists] "/workspace/src/Vitis-AI-Quantizer/vai_q_pytorch/example/vit_b_16/quantize_result/*.xmodel" doesn't exist.
*** Check failure stack trace: ***
This program has crashed!
Aborted (core dumped)

```

Please advise how to workaround this? or how to offload parts of this to the CPU?

Alexiazzf, 2023-02-20:
Hi. Quantization of torch.int64 and torch.int32 is not supported yet. We only support float32/double/float16 quantization. Your error is beacause XIR does not support multi-outputs op. You can use QuantStub() and DeQuantStub() in your model to offload parts of this to the CPU.(You can see more details in [README.md](https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_quantizer/vai_q_pytorch/README.md)

anirudhitagi, 2023-02-20:
Hi @Alexiazzf, 
Thank you for your response and the suggestion. Can you please help me understand how to add the QuantStub() and DeQuantStub() to pretrained model? for example: I am working with the pretrained model from the torchvision module : 
```
from pytorch_nndct.apis import torch_quantizer
import torch
import torchvision
from torchvision.models import vit_b_16
model = vit_b_16(weights='IMAGENET1K_V1').cpu()
```

How do I add QuantStub() and DeQuantStub() function here? 
Any help would be greatly appreciated!

Alexiazzf, 2023-02-21:
Hi @anirudhitagi ,
You could add QuantStub() and DeQuantStub() into your vit_b_16 model. You can find our example in  https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_quantizer/vai_q_pytorch/example/resnet18_qat.py#L251. And you can see the instruction in [README.md](https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_quantizer/vai_q_pytorch/README.md) which can answer your questions exactly.

anirudhitagi, 2023-03-02:
Hi @Alexiazzf ,
sorry to follow up regarding the same doubt, but I am struggling to add QuantStub() and DeQuantStub() functions in the vit_b_16 model. 

In the example that you have shared above, it looks like first the entire resnet model is coded manually and then the QuantStub() and DeQuantStub() functions are inserted. is there another way to insert these functions directly into the vit_b_16 model?

Alexiazzf, 2023-03-06:
Hi @anirudhitagi ,

We only support insert QuantStub() and DeQuantStub() manually into the model. If you do not add QuantStub() and DeQuantStub(), you will quantize the whole model.

Truthonlyone, 2023-05-10:
> Hi. Quantization of torch.int64 and torch.int32 is not supported yet. We only support float32/double/float16 quantization. Your error is beacause XIR does not support multi-outputs op. You can use [...]

Hey,
  I use vitis-ai-pytorch to quantize my model, and the below warning appears:
  "[VAIQ_WARN]: The tensor type of  Model::input_0 is torch.float16. Only support float32/double quantization."
  My Vitis AI version is VitisAI2.5 GPU. Is the GPU version dose not support float16 quantization?
  Then I print the "dtype" of the model, and it shows "torch.float32". Can this proves that the precision of the model is float32? If not, would you mind tell me how can I judge the accuracy of my model?
Thx~

closed 2023-06-26 by niuxjxlnx