Xilinx/Vitis-AI #1178: Invalid version: '3.0.0-a44284e-torch1.12.1' (package: pytorch-nndct)

ZQ-zz, 2023-02-20:
When I use 'pip install -v -e .' to package my code (which depends on pytorch-nndct)，errors like below
```
      File "/home/zq/anaconda3/envs/sxcv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2631, in hashcmp
        self.parsed_version,
      File "/home/zq/anaconda3/envs/sxcv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2685, in parsed_version
        raise packaging.version.InvalidVersion(f"{str(ex)} {info}") from None
    pkg_resources.extern.packaging.version.InvalidVersion: Invalid version: '3.0.0-a44284e-torch1.12.1' (package: pytorch-nndct)
    error: subprocess-exited-with-error
```
it seems '3.0.0-a44284e-torch1.12.1' is a invalid version, how to fix it ?

Alexiazzf, 2023-02-20:
Hi. I cannot get your error with VAI 3.0 and pytorch 1.12.1. Does it work with installing pytorch-nndct by `python setup.py develop` or `python setup.py install`?

ZQ-zz, 2023-02-21:
Hi @Alexiazzf , I use ‘conda install pytorch_nndct=3.0.0 xcompiler -c pytorch -c conda-forge -c file:///local/zq/vitis/3.0/conda-channel’ (https://www.xilinx.com/bin/public/openDownload?filename=conda-channel-3.0.tar.gz)
pip list info:
```
...
pytorch-nndct      3.0.0-a44284e-torch1.12.1
...
torch              1.12.1
...
```
Could you show your pip list info?

Alexiazzf, 2023-02-22:
Hi @ZQ-zz , My pip info is the same as yours. Maybe your could delete this file `nndct/pytorch_binding/version.txt` then try again. Or you could reinstall pytorch-nndct.