Xilinx/Vitis-AI #1181: docker_run.sh doesn't work in a path whit spaces

barbiomalefico, 2023-02-21:
If you clone Vitis AI in a folder where the full path contains spaces, the docker_run.sh script fails. 
For example:
/home/user/Folder with spaces/Vitis/Vitis-AI

janifer112x, 2023-03-01:
Hi @barbiomalefico，
please avoid of using home directory as the workspace of launching Vitis AI docker, as it might have some permission problem with mounting directory from the host into container.
also folders with space is not support as you mentioned

barbiomalefico, 2023-03-01:
> 

It's not a problem related to the home directory or permissions. **It's a problem related to a path that includes spaces**.

mcbullisisu, 2023-03-15:
I can't comment on the concern about volume mounting the repository cloned to the home directory into the container.

The issue with spaces in the path to the repository could probably be fixed with some simple quoting around the volume mount in the docker run command  `-v "$DOCKER_RUN_DIR" :/vitis_ai_home \`