Xilinx/Vitis-AI #1183: Error encountered deploying .xModel in ZCU104

shoayi, 2023-02-24:
Hello! I used vitis-ai 2.5 to compile my model successfully. 
[image]
but when i run python code to call model, this error occur.
[image]
There is my code.
[ZYNQ_inference.zip](https://github.com/Xilinx/Vitis-AI/files/10819611/ZYNQ_inference.zip)

lishixlnx, 2023-03-28:
you attached the wrong code.
the code attached doesn't match with the error log.
for example, line number is 71 in the file but there is "line 82" in log.

qianglin-xlnx, 2023-06-25:
Closing since no activity for more than 2 months, please reopen if you still have question, thanks

closed 2023-06-25 by qianglin-xlnx