Xilinx/Vitis-AI #1187: [VAILIB_CPU_RUNNER_OPEN_LIB_ERROR] libvart_op_imp_strided_slice-fix.so: cannot open shared object file

j0r0123, 2023-03-01:
Hello everyone,

I want to implement my PyTorch model using Vitis AI 3.0. My model contains code that splits a tensor into halves. For the sake of simplicity, I created a minimal working example:

```
class myModel(nn.Module):
    def __init__(self):
        super(myModel, self).__init__()

    def forward(self, x: torch.Tensor):
        shape_half = 144
        xi = x[:, :shape_half, :]
        return xi
``` 

The xir compiler uses the strided_slice-fix function to implement this. The graph of the minimal working example looks like the following:
[image]

The problem occurs when I try to deploy this model. I get the following error message:

```
F0301 13:57:30.340754  1622 op_imp.cpp:110] [UNILOG][FATAL][VAILIB_CPU_RUNNER_OPEN_LIB_ERROR][dlopen can not open lib!]  lib=libvart_op_imp_strided_slice-fix.so;error=libvart_op_imp_strided_slice-fix.so: cannot open shared object file: No such file or directory;op=xir::Op{name = myModel__myModel_23, type = strided_slice-fix}
```

It seems like the strided_slice-fix function is not implemented. I checked the vai_library and can only find an implementation for strided_slice but not for strided_slice-fix. 

Is there any workaround or fix for this problem? Will this function be implemented in a future version of Vitis AI?
Thanks for your help!

MingyangZhang77, 2023-04-11:
Hi @j0r0123 ,
I'm facing the same problem when trying to run my own model on a DPU. Did you find any workaround?

j0r0123, 2023-04-11:
Hi @MingyangZhang77,

I used the custom op work flow in order to implement the function myself. The strided-slice (without fix-point) implementation served me as a reference point for that. 

I hope that helps.

omgitsmj24, 2023-05-30:
I'm facing the same problem, does that mean you have to write a custom op for a op that is already present in the /usr/lib, and write a pure python implementation and copy it as "cp vart_op_imp/ /usr/lib/python3.9/site-packages/ -r"?

This is one more thing I'd like to add about this problem is that, the output of the inspect function for this op say that it is not assigned to DPU. But when I quantize, compile and inspect using netron it say it is assigned to DPU without any problem until I try to run_op using xdputil that is when I find out there is not slice stride present in /usr/lib.

lishixlnx, 2023-06-18:
@j0r0123 's answer is correct. 
this need be implemented by custom op.

@omgitsmj24 you mentioned compile and inspect by netron and found it is assigned to DPU,
can you please paste your compiled model here for reference?
thanks