Xilinx/Vitis-AI #1192: Can't download Board Image of KV260 in Vitis AI v3.0

hill19850213, 2023-03-03:
Hi,

I can't download kv260 image from below link
[https://xilinx-ax-dl.entitlenow.com/dl/ul/2023/01/06/R210770828/xilinx-kv260-dpu-v2022.2-v3.0.0.img.gz?hash=QsYjeGyUk6zkwxtph6Ka8Q&expires=1677887815&filename=xilinx-kv260-dpu-v2022.2-v3.0.0.img.gz](url)

Is this link broken?

eerobert, 2023-03-04:
hi

https://xilinx.github.io/Vitis-AI/docs/board_setup/board_setup_mpsoc.html

please try the above link

hill19850213, 2023-03-04:
It's the same situation for KV260, but the link of zcu102 and zcu104 are workable.

eerobert, 2023-03-04:
the link works for me.. but require xilinx login

https://www.xilinx.com/member/forms/download/design-license-xef.html?filename=xilinx-kv260-dpu-v2022.2-v3.0.0.img.gz

i also attach the actual link from the board_setup_mpsoc page for reference

hill19850213, 2023-03-06:
I have done the login and download this image, but chrome is still shown network error as below
[image]

hill19850213, 2023-03-06:
Finally, I use vpn to download this file, it is success.

closed 2023-03-27 by qianglin-xlnx