Xilinx/Vitis-AI #1195: vai_c_xirv compiler did not take any modified arch.json, but I need to modify fingerprint.

mdrz7, 2023-03-06:
Hi,
I encounter an error after many testing. I have a KV260 board. I know have to change the fingerprint for the compiler to get the correct xmodel file.
I tried:
1. Add a new arch.json under docker workspace. Error with: Unable to open file /workspace/arch.json
2. Add a new arch.json under a new arch folder workspace. Error with: Unable to open file /workspace/arch/arch.json
3. Modified the default kv260 json under /opt/vitis_ai/compiler/arch/DPUCZDX8G/KV260/arch.json with new fingerprint. After I modified this file, errored with: Unable to open file /opt/vitis_ai/compiler/arch/DPUCZDX8G/KV260/arch.json
4. Use the default kv260 arch.json in the /opt/vitis_ai/compiler/arch/DPUCZDX8G/KV260/arch.json without touching it. Compiled successfully!!! But when I put the xmodel to the board. This error shows up: 
`W0306 15:21:42.560640  8860 dpu_runner_base_imp.cpp:733] CHECK fingerprint fail! model_fingerprint 0x1000020f6014407 is un-matched with actual dpu_fingerprint 0x101000056010407. Please re-compile xmodel with dpu_fingerprint 0x101000056010407 and try again.`

It's kinda like a loop. Without correct fingerprint arch.json file -> no correct xmodel file -> edit the arch.json file -> does not take the JSON to compile xmodel -> loop

I tried Docker Image Version: 2.5.0.1260   (CPU) and Docker Image Version: 2.0.0.1103   (CPU), both having the same problem
kv260 system image: xilinx-kv260-dpu-v2022.2-v3.0.0.img

Please let me know what to do.  Many thanks,

LogicTronix, 2023-03-07:
@mdrz7 can you post here "full command you ran for Compilation", so that we can see "if you have/not misplace JSON file, models name, location etc" while running compilation.
And your KV260 image is based on "Vitis AI 3.0" while your "Vitis AI CPU docker seems to be 2.5 or 2.0".

mdrz7, 2023-03-07:
> @mdrz7 can you post here "full command you ran for Compilation", so that we can see "if you have/not misplace JSON file, models name, location etc" while running compilation.
> And your KV260 image is based on "Vitis AI 3.0" while your "Vitis AI CPU docker seems to be 2.5 or 2.0".

HI! Thank you for your reply. After testing my compilation cmd this is what I figure out:

Working cmd, generated CNN_KV260.xmodel immediately. Using the default /opt/.../KV260/arch.json came with docker image, but unable to run on the board. (CHECK fingerprint fail)
`vai_c_xir --xmodel ./build/quant_model/cnn_int.xmodel --arch /opt/vitis_ai/compiler/arch/dpuczdx8g/kv260/arch.json --net_name cnn_kv260 --output_dir ./build/compiled_model`
[image]

Compile failed cmd, the only difference is I point to another arch.json file that I modified the fingerprint in it.
`vai_c_xir --xmodel ./build/quant_model/cnn_int.xmodel --arch ./build/arch.json --net_name cnn_kv260 --output_dir ./build/compiled_model`
[image]

mdrz7, 2023-03-07:
I managed to figure out the reason. That error basically means my arch.json is having the wrong "syntax".
I was using {fingerprint: "0x10000xxxxxxxx"} actuelly is {"fingerprint": "0x1010000xxxxxxx"}. I missed the "" for the fingerprint. Hope this helps someone...