Xilinx/Vitis-AI #1196: How to test deployable model in QAT?

LorenzoSun-V, 2023-03-09:
In [new qat example](https://github.com/Xilinx/Vitis-AI/blob/master/examples/vai_quantizer/pytorch/resnet18_qat.py#L650), I found `mode` has been changed to `['train', 'deploy']` in VitisAI3.0 from `['train', 'deploy', 'test']` in 2.5 .
So I load `deployable.pth` directly after building QAT model and infer on my dataset, but the results are lower than results of validation in training. 
Is there any way to get the depolyable model metrics in 3.0 ?

yuwang-xlnx, 2023-03-23:
Hi @LorenzoSun-V,

> So I load deployable.pth directly after building QAT model and infer on my dataset, but the results are lower than results of validation in training.

A deployable model can be obtained either by `to_deployable()` or `deployable_model()`. You are not able to generate a deployable model by yourself. See https://github.com/Xilinx/Vitis-AI/blob/v3.0/src/vai_quantizer/vai_q_pytorch/example/resnet18_qat.py#L643

LorenzoSun-V, 2023-03-24:
Thanks for your reply.
My purpose is that testing the accuracy of the model quantized by QAT on GPU and aligning the accuracy of the model on the board, whereas the `test` mode included in QAT is removed in VitisAI3.0. So I want to load the weights of `deployable.pth` generated by `to_deployable()` or `deployable_model()` into my model and infer it on my dataset.
Do you have any idea about testing the accuracy of model quantized by QAT on GPU?

yuwang-xlnx, 2023-03-24:
Hi @LorenzoSun-V,

You can do it like this:

```
qat_processor = QatProcessor(model, inputs)
# 'output_dir' is the directory you passed to 'to_deployable()' when you call it
deployable_model = qat_processor.deployable_model(output_dir)
validate(val_loader, deployable_model)
```

LorenzoSun-V, 2023-03-27:
Thanks @yuwang-xlnx .
I'll try that and feedback later.

LorenzoSun-V, 2023-03-28:
It works well. Thanks.

closed 2023-03-28 by LorenzoSun-V