Xilinx/Vitis-AI #1197: Does quantizer.quant_model behave like the model on the board?

alewarne, 2023-03-10:
The prediction method of the `quantizer.quant_model` seems to perform the quantization in a software simulation, i.e. during the forward pass the layer weights are quantized and then the output is computed and propagated. I found, however, that the prediction performance on the board is not similar to the `quantizer.quant_model`. Is this behavior correct? If so, could you tell me what the difference between the `quantizer.quant_model` and the final model on the board is?

Alexiazzf, 2023-03-23:
Hi @alewarne ,

The quantizer.quant_model simulated the accuracy of model on the board rather than prediction performance. So this behavior is correct.

alewarne, 2023-03-23:
I don't quite get it, sorry. The accuracy of the model on the board and the prediction performance are the same in my view. I don't mean performance in terms of runtime, only in terms of classification performance (=Accuracy).

Alexiazzf, 2023-06-26:
Hi @alewarne ,

The accuracy of the model on the board should be similar to the quantizer.quant_model. You could use newest version of VAI in case we have fixed your issue already. If it does not help, you could give more details about your issue.