Xilinx/Vitis-AI #1200: vai_c_xir compilation ends with "Error without ID assigned occur"

ReinForce-II, 2023-03-14:
I tried to compile a model, ends with an "Error without ID assigned occur".
Is there a way to debug the error?
$ vai_c_xir -x quantize_result/CModel_int.xmodel -a arch.json -o compile_result
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
[UNILOG][INFO] Graph name: CModel, with op num: 113
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][ERROR_SAMPLE][Error without ID assigned occur!]
*** Check failure stack trace: ***
Aborted (core dumped)

[quantize_result.zip](https://github.com/Xilinx/Vitis-AI/files/10964697/quantize_result.zip)

ajithvallabai, 2023-03-14:
Did you took arch.json from /opt/vitis_ai/compiler/arch/ folder eg:./opt/vitis_ai/compiler/arch/DPUCV2DX8G/VEK280/arch.json ?

ReinForce-II, 2023-03-14:
> Did you took arch.json from /opt/vitis_ai/compiler/arch/ folder eg:./opt/vitis_ai/compiler/arch/DPUCV2DX8G/VEK280/arch.json ?

Yes, content of arch.json:
```
{"fingerprint":"0x101000016010407"}
```