Xilinx/Vitis-AI #1201: Compilation error "quantizing is out of range" in TensorFlow2 custom model

ida255, 2023-03-20:
Thanks for releasing Vitis-AI 3.0!

I'm in the process of porting our custom CNN model for the KV260.
I have already been able to generate a quantized quintized.h5 model file.
When I compile (vai_c_tensorflow2) this model, I get the following error.

`[UNILOG][FATAL][XCOM_FRONTEND_QUANT_ATTR_OUT_OF_RANGE][The shiftbit for quantizing is out of range, that the range is restricted by target DPU.] Invalid shift_bias(-10) for xir::Op{name = quant_dec_crop1_2x2_, type = depthwise-conv2d-fix}, where other shift info is i_pos(5), w_pos(6), b_pos(21), o_pos(4).`

The "dec_crop1_2x2" layer is Conv2D added to the network in place of Cropping2D.
Therefore, the kernel is [[0,0],[0,1]] in order to remove the top and left lines.

Below is information about the layers.
dilations: 1x1
group: 256
kernel_shape: 2,2
pads: 0,0,0,0
stride: 1,1

Is there any way to resolve this compile error?

BR

ida255, 2023-03-21:
The error message says shift_bias -10, but I don't know how to change the shift_bias.
Also, I'm not sure about the valid shift_bias range when using the DPU.
I would be happy to receive any comments.

quentonh, 2023-03-31:
@ida255 Can you check your quantization log to see if the shift_bias has been adjusted based on the DPU compiler limitations?  You should check the log for this message:

https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_quantizer/vai_q_tensorflow2.x/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_refine_transforms.py#L359  

It would be great if you could post your quantized model.

Starting here:

https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_quantizer/vai_q_tensorflow2.x/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_refine_transforms.py#L309

What it appears should happen is that a new_sb of -9 should be assigned and that the result will be that a new b_pos will be written back to the layer by the quantizer.

# From your error message:
i_pos(5), w_pos(6), b_pos(21), o_pos(4)

# Calc shift_cut
shift_cut = wpos + ipos - opos
shift_cut = 6 + 5 - 4
shift_cut = 7

# Calc min_sb
min_sb = min(0, -(24 - (8 + shift_cut)))
min_sb = min(0, -(24 - (8 + 7)))
min_sb = min(0, -9)
min_sb = -9

# Assign max_sb
max_sb = 16

#Calc shift_bias
shift_bias = wpos + ipos - bpos
shift_bias = 6 + 5 - 21
shift_bias = -10

# Checks if -10 < -9, assigns new_sb = -9
new_sb = None
      if shift_bias < min_sb:
        new_sb = min_sb
      elif shift_bias > max_sb:
        new_sb = max_sb

# Apparently this code should override the shift_bias, but doesn’t???
if new_sb is not None:
        new_bpos = wpos + ipos - new_sb
        self.set_pos(layer, 'b', new_bpos)
        logger.debug('Shift bias of layer {} is {}. It exceeds range [{}, {}]. '
                     'Modify bpos from {} to {}.'.format(
                         layer.name, int(shift_bias), int(min_sb), int(max_sb),
                         int(bpos), int(new_bpos)))

new_bpos = 6 + 5 - (-9)
new_bpos = 20

renjiangyong, 2023-04-03:
Hi, @ida255  It would be great if you could provide your origin model, so We can repeat this issue, So We can fix it. Thank you!

ida255, 2023-04-03:
Hi everyone.
Thanks for your comments.
Due to confidentiality, I cannot share the model.

Overriding shift_bias might be a try.

I also made an inquiry on the Xilinx Forum with the same content.
https://support.xilinx.com/s/question/0D74U000007LvRgSAK/detail

The answer was that it is necessary to normalize the input data.
The model also needs to be retrained to match this data normalization.

renjiangyong, 2023-04-03:
Hi @ida255 ,  you can try to  normalize the input data first. if you cann't share the model with confidentiality, Thank you!

closed 2023-04-05 by ida255