Xilinx/Vitis-AI #1202: Vitis-AI profiler support for non ZOCL runtime

MaistoV, 2023-03-20:
Does Vitis-AI profiler support:
- DPUCZDX8G device for ZCU102?
- Bare-metal runtime, just dpu.ko driver, no ZOCL (Zynq Open CL) runtime?

Have there been any changes between v2.0 and v2.5?

Thank you for the help.

jileilxinlix, 2023-04-03:
hi @MaistoV :
 1.Does Vitis-AI profiler support:
    we support simple linux runtime without ZOCL(this same to Vivado flow)
2.Have there been any changes between v2.0 and v2.5?
   There is no changes from user perspective

MaistoV, 2023-04-06:
I built the kernel and rootfs with Petalinux 2021.2 and enabled Kernel tracing as in https://github.com/Xilinx/Vitis-AI-Tutorials/blob/1.4/Design_Tutorials/16-profiler_introduction/README.md . As a matter of fact, I can use kprobes and uprobes directly.

I am currently using design synthesized from Vivado flow from TRD from Vitis-AI 2.0 ( https://github.com/Xilinx/Vitis-AI/tree/v2.0/dsa/DPU-TRD/prj/Vivado ). I succesfully ran multiple CNNs on different designs with several DPU configurations and one or more DPU cores.

When I run vaitrace I get the following warning and no computing units are found:
- WARNING:root:Cannot open 'zyxclmm_drm/ip_layout'

I checked sysfs and none of the following devices, used by the tracer python scripts, can be found:
- '/sys/devices/platform/amba/amba:zyxclmm_drm/ip_layout',
- '/sys/devices/platform/axi/axi:zyxclmm_drm/ip_layout',
- '/sys/devices/platform/amba_pl@0/amba_pl@0:zyxclmm_drm/ip_layout'

Edocit, 2023-04-08:
I reported exactly the same issue in my post here

 https://support.xilinx.com/s/question/0D74U000007Lvu4SAC/detail?language=en_US&fromEmail=1&s1oid=00D2E000000nHq7&s1nid=0DB2E000000XdtH&s1uid=0052E00000N3TnZ&s1ext=0&emkind=chatterCommentNotification&emtm=1679880861901 

I also provided additional information. Please help us, I'm stuck waiting for a solution since I need to use the tracer.

jileilxinlix, 2023-04-10:
@Edocit 
hello @Edocit  I check your issue in our board and user vart version3.0:
[image]
The version:
[image]

jileilxinlix, 2023-04-10:
> I built the kernel and rootfs with Petalinux 2021.2 and enabled Kernel tracing as in https://github.com/Xilinx/Vitis-AI-Tutorials/blob/1.4/Design_Tutorials/16-profiler_introduction/README.md . As a [...]

hi @MaistoV  
Maybe this warning has no effect on vaitrace, Whether the vaitrace works correctly?

Edocit, 2023-04-12:
Can you detail the procedure you used to test with VAI 3.0. Did you used Vivado 2022.2 ? Did you create the project from BSP or from --template zynqMP ? Have you made something particular in the Vivado design ? I still have issues with petalinux 2022.2 (at boot it can't bring up ethernet and then when running the patch for the PMIC it says the PMIC is busy and can't talk to it via I2C). Thank you, Edoardo

MaistoV, 2023-04-12:
> hi @MaistoV Maybe this warning has no effect on vaitrace, Whether the vaitrace works correctly?

It has effect, as a matter of fact, the sources of vaitrace are available [here](https://github.com/Xilinx/Vitis-AI/tree/v2.0/tools/Vitis-AI-Runtime/VART/vart/trace) and this warning is generated from [hwInfo.py](https://github.com/Xilinx/Vitis-AI/blob/d02dcb6041663dbc7ecbc0c6af9fafa087a789de/tools/Vitis-AI-Runtime/VART/vart/trace/vaitrace/tracer/hwInfo.py#L147)

jileilxinlix, 2023-04-14:
> It has effect, as a matter of fact, the sources of vaitrace are available [here](https://github.com/Xilinx/Vitis-AI/tree/v2.0/tools/Vitis-AI-Runtime/VART/vart/trace) and this warning is generated [...]

@MaistoV  :
1. After we double check the code, this warning not effect vaitrace, because we not use hare ware info data in vaitrace version2.0.This hwinfo data just collect. But we still thank your for your commit, for vivado flow we should not use this method collect hwinfo data. 
 In addition, do you run vaitrace success except this warnning?

BR
Jilei

jileilxinlix, 2023-04-14:
> Hi, what was the process you did to build and successfully test the platform ? Petalinux 2022.2 and Vivado 2022.2 and the master branch of Vitis-AI on github. I create a project using the --template zynqMP option, then using petalinux-config --get-hw-description I imported the Vivado platform (the XSA) and changed in the DTG settings the platform to be zcu104-revc. Then added the DPU kernel module after petalinux-config -c kernel and added the needed stuff in order to use vaitrace according to https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Getting-Started-with-the-Vitis-AI-Profiler Then copied the 2 recipies and commented the line to disable vitis flow in vart.bb Added vitis-ai and vitis-ai-dev in the menu prompted by petalinux-config -c rootfs and enabled the packagegroup-selfhosted After petalinux-build and package when the board boots I faced 2 issues: the first one is that I can no more succesfully run the dpu_optimize_script to adjust PMIC settings and then dnf (that I added in the rootfs before build) does not work with output message 'var/lock exist' message. I ran ip a command and I saw network was no more able to setup at boot and when I execute if up eth0 the interface becomes active but can't associate a DNS. It seems something missing in the rootfs, and I think it is something related to tmp filesystem directories. Do you have some hint about it ? the Vivado platform is a simple design only including a dual core DPU. However, I can provide pictures about it if needed or if this can be useful for you to help me ? Thank you, Edoardo Il giorno lun 10 apr 2023 alle 08:22 jileilxinlix ***@***.***> ha scritto:
> […](#)

hi @Edocit :
Our image build and vitis-ai install flow:https://github.com/Xilinx/Vitis-AI/blob/master/dpu/ref_design_docs/README_DPUCZ_Vitis.md ,we use vitis flow. 
If you want to use vivado flow, you should follow:https://github.com/Xilinx/Vitis-AI/blob/master/dpu/ref_design_docs/README_DPUCZ_Vivado.md 

I will recurrence your issue in my environment follow your step, but i need more time.

MaistoV, 2023-04-26:
> @MaistoV :
>
> 1. After we double check the code, this warning not effect vaitrace, because we not use hare ware info data in vaitrace version2.0.This hwinfo data just collect. But we still thank your [...]

No, I get this error and no output files are produced:

Traceback (most recent call last):
  File "/usr/bin/vaitrace", line 186, in <module>
    main(*parseCmdLine())
  File "/usr/bin/vaitrace", line 181, in main
    vaitraceWriter.write(options)
  File "/usr/bin/xlnx/vaitrace/vaitraceWriter.py", line 83, in write
    writer.vtf.convert.xat_to_vtf(rawData)
  File "/usr/bin/xlnx/vaitrace/writer/vtf/convert.py", line 331, in xat_to_vtf
    convert_dpu(xat.get('vart'))
  File "/usr/bin/xlnx/vaitrace/writer/vtf/convert.py", line 140, in convert_dpu
    dpu_profile_summary = dpu_parser.get_dpu_profile_summary()
  File "/usr/bin/xlnx/vaitrace/writer/parser/vart_parser.py", line 351, in get_dpu_profile_summary
    perf = workload * batch / avg_t * 1000  # to GOP/ms
ZeroDivisionError: float division by zero

The cause, as described above, is that no computing units are found, so the variable avg_t holds a zero value.

bvleeuwentue, 2023-06-08:
Did anyone find a solution to this issue? 
I get a similar Error when using the Vivado flow for 2022.2 and VitisAI 3.0.
```
Traceback (most recent call last):
  File "/usr/bin/vaitrace", line 195, in <module>
    main(*parseCmdLine())
  File "/usr/bin/vaitrace", line 190, in main
    vaitraceWriter.write(options)
  File "/usr/bin/xlnx/vaitrace/vaitraceWriter.py", line 94, in write
    writer.vtf.convert.xat_to_vtf(rawData, options)
  File "/usr/bin/xlnx/vaitrace/writer/vtf/convert.py", line 327, in xat_to_vtf
    convert_dpu(xat.get('vart'), xat.get('hwInfo'), options)
  File "/usr/bin/xlnx/vaitrace/writer/vtf/convert.py", line 158, in convert_dpu
    s, e = toVTF(event)
  File "/usr/bin/xlnx/vaitrace/writer/vtf/convert.py", line 95, in toVTF
    efficiency = round(workload / time_dur, 2)
ZeroDivisionError: float division by zero
```
I followed the steps described [here](https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Installing-the-Vitis-AI-Profiler) but I did not get it to work. 

The DPU version:
[image]

Thanks in advance

jileilxinlix, 2023-06-09:
@bvleeuwentue 
This error will be encounter in vivado flow, we will fix it in version-3.01.

Shreyas-NR, 2024-01-09:
Hi @qianglin-xlnx , @jileilamd @bvleeuwentue 
Is this issue fixed? When can I expect the version 3.01 to be released?
What steps I need to do to get this upgrade?
Could you please help me with the above?

-Shreyas

jileilamd, 2024-01-09:
@Shreyas-NR  you could use lasted code to fix this error, the fix has been included in lasted version

Shreyas-NR, 2024-01-09:
Hi @jileilamd ,
Thanks for the quick response,
Could you please direct me to the latest code?

Shreyas-NR, 2024-01-09:
Hi @jileilamd ,
I'm doing Vivado flow, petalinux 2022.2 Vitis AI 3.0

[image]

[image]

Could you please help me which code to change and its location?

Looking forward to hearing from you.

jileilamd, 2024-01-10:
hi @Shreyas-NR : you could use https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_runtime/vart/trace/src/time.cpp this file replace your same file, and build again. please have a try , if any problem please feel free to let me know

Shreyas-NR, 2024-01-10:
Hi @jileilamd ,
Thank you so much for your replies. It is helping me a lot. 

I updated the Vitis AI on my board with the latest Vitis AI 3.5.
For some reason vaitrace didn't generate the .run_summary file

```
root@xilinx-kv260-starterkit-20222:~/app/samples/python# python3 -m vaitrace_py resnet50_mt_custom.py 2
INFO:root:VART will run xmodel in [NORMAL] mode
Analyzing symbol tables...
61 / 61
81 / 81
6 / 6
45 / 45
3 / 3
INFO:root:vaitrace compile python code: resnet50_mt_custom.py
INFO:root:vaitrace exec poython code: resnet50_mt_custom.py
Thread: 281473665454416 , Top[0] 280 0.144820 "grey fox, gray fox, Urocyon cinereoargenteus,",
Thread: 281473657000272 , Top[0] 945 0.779940 "bell pepper,",
Thread: 281473665454416 , Top[1] 279 0.112786 "Arctic fox, white fox, Alopex lagopus,",
Thread: 281473657000272 , Top[1] 941 0.286924 "acorn squash,",
Thread: 281473665454416 , Top[2] 269 0.064263 "timber wolf, grey wolf, gray wolf, Canis lupus,",
Thread: 281473657000272 , Top[2] 943 0.223457 "cucumber, cuke,",
Thread: 281473665454416 , Top[3] 278 0.056712 "kit fox, Vulpes macrotis,",
Thread: 281473657000272 , Top[3] 952 0.119608 "fig,",
Thread: 281473665454416 , Top[4] 272 0.053276 "coyote, prairie wolf, brush wolf, Canis latrans,",
Thread: 281473657000272 , Top[4] 939 0.087507 "zucchini, courgette,",
FPS=8.89, total frames = 1.00 , time=0.112459 seconds
APM Stop Collecting
WARNING:root:Cannot open 'zyxclmm_drm/ip_layout'
INFO:root:Generating VTF
INFO:root:Processing xmodel information
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_AvgPool2d_avgpool__8077_fix|8202529256]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_QuantStub_quant_stub__input_1_upload_0|0]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Conv2d_conv1__input_2|238637056]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer1__Bottleneck_0__Conv2d_conv1__input_5|25890816]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer1__Bottleneck_0__Conv2d_conv2__input_8|231411712]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer1__Bottleneck_0__Conv2d_conv3__input_11|103563264]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer1__Bottleneck_0__Add_skip_add__input_13|105168896]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer1__Bottleneck_1__Conv2d_conv1__input_15|102961152]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer1__Bottleneck_1__Conv2d_conv2__input_18|231411712]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer1__Bottleneck_1__Add_skip_add__input_22|105168896]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer1__Bottleneck_2__Conv2d_conv1__input_24|102961152]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer1__Bottleneck_2__Conv2d_conv2__input_27|231411712]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer1__Bottleneck_2__Add_skip_add__input_31|105168896]
INFO:root:SubGraph index duplication: [subgraph_fake_downsample_0|0]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_0__Conv2d_conv1__input_33|205922304]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_0__Conv2d_conv2__input_36|231311360]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_0__Conv2d_conv3__input_39|103161856]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_0__Add_skip_add__input_41|206725120]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_1__Conv2d_conv1__input_43|102860800]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_1__Conv2d_conv2__input_46|231311360]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_1__Add_skip_add__input_50|103964672]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_2__Conv2d_conv1__input_52|102860800]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_2__Conv2d_conv2__input_55|231311360]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_2__Add_skip_add__input_59|103964672]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_3__Conv2d_conv1__input_61|102860800]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_3__Conv2d_conv2__input_64|231311360]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer2__Bottleneck_3__Add_skip_add__input_68|103964672]
INFO:root:SubGraph index duplication: [subgraph_fake_downsample_1|0]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer3__Bottleneck_0__Conv2d_conv1__input_70|205721600]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer3__Bottleneck_0__Conv2d_conv2__input_73|231261184]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer3__Bottleneck_0__Conv2d_conv3__input_76|102961152]
INFO:root:SubGraph index duplication: [subgraph_ResNet__ResNet_Sequential_layer3__Bottleneck_0__Add_skip_add__input_78|206123008]
[... cut ...]
             : also emulators implemented with same interface to make debuging eaiser.
```

```
root@xilinx-kv260-starterkit-20222:~/app/samples/python# ls -l
total 188
-rw-r--r-- 1 root root    724 Jan  9 22:06 profile_summary.csv
-rw-r--r-- 1 1122 1122  12110 Jan  9 01:04 resnet50_mt_custom.py
-rw-r--r-- 1 root root    585 Jan  9 22:06 vart_trace.csv
-rw-r--r-- 1 root root 142273 Jan  9 22:06 vitis_ai_profile.csv
-rw-r--r-- 1 1122 1122  25674 Jan  8 21:39 words.txt
root@xilinx-kv260-starterkit-20222:~/app/samples/python#

```

Can you help me with this issue?

jileilamd, 2024-01-10:
hi @Shreyas-NR : you could add -d  your command and this will output  vart.trace. please send this file to me , I'll debug it.

Shreyas-NR, 2024-01-10:
Hi @jileilamd,
Thanks for getting back

Here is the log of the execution. 

```
root@xilinx-kv260-starterkit-20222:~/app/samples/python# python3 -m vaitrace_py -d  resnet50_mt_custom.py 2

DEBUG:root:Command line args:
DEBUG:root:Namespace(cmd=['resnet50_mt_custom.py', '2'], config='', debug=True, traceSaveTo=None, timeout=None, showversion=False, bypass=False, python=True, va=False, xat=False, txt=False, json=False)
DEBUG:root:Use boot as trace clock
INFO:root:VART will run xmodel in [NORMAL] mode
DEBUG:root:Control Options:
DEBUG:root:{'cmd': ['resnet50_mt_custom.py', '2'], 'xat': {'compress': True, 'filename': '/home/root/app/samples/python/resnet50_mt_cus.xat'}, 'config': None, 'timeout': 60, 'debug': True, 'traceClock': 'boot', 'x86_tsc_khz': 0, 'platform': {'idcode': '0x4724093', 'fmlyid': '0x23', 'sub_fmlyid': '0x4', 'devid': '0x24', 'model': 'xlnx,zocl', 'os': 'Linux', 'release': ['5', '15'], 'machine': 'aarch64'}, 'runmode': 'normal'}
DEBUG:root:Preparing [hwInfo]...
DEBUG:root:Compatible [hwInfo]: True
DEBUG:root:Preparing [function]...
DEBUG:root:Compatible [function]: True
Analyzing symbol tables...
61 / 61
81 / 81
6 / 6
45 / 45
3 / 3
DEBUG:root:Preparing [sched]...
DEBUG:root:Preparing [xapm]...
DEBUG:root:Compatible [xapm]: True
DEBUG:root:Preparing [nmu]...
DEBUG:root:Compatible [nmu]: True
DEBUG:root:Preparing [cmd]...
DEBUG:root:Compatible [cmd]: True
DEBUG:root:Preparing [vart]...
DEBUG:root:Compatible [vart]: True
DEBUG:root:Preparing [pyfunc]...
DEBUG:root:Compatible [pyfunc]: True
DEBUG:root:Preparing [power]...
DEBUG:root:boot
DEBUG:root:### uprobe p:_fun_3296584_entry /usr/lib/libvitis_ai_library-facedetectrecog.so:0xe2d0
DEBUG:root:### uprobe p:_fun_7663178_entry /usr/lib/libvitis_ai_library-facedetectrecog.so:0xe7c0
DEBUG:root:### uprobe p:_fun_5011292_entry /usr/lib/libvitis_ai_library-mnistclassification.so:0x3450
DEBUG:root:### uprobe p:_fun_471863_entry /usr/lib/libvitis_ai_library-textmountain.so:0x3a50
DEBUG:root:### uprobe p:_fun_6052259_entry /usr/lib/libvitis_ai_library-textmountain.so:0x3ff0
DEBUG:root:### uprobe p:_fun_2015311_entry /usr/lib/libvitis_ai_library-fusion_cnn.so:0xd7f0
DEBUG:root:### uprobe p:_fun_806461_entry /usr/lib/libvitis_ai_library-facelandmark.so:0x3460
DEBUG:root:### uprobe p:_fun_3563505_entry /usr/lib/libvitis_ai_library-facelandmark.so:0x2e30
DEBUG:root:### uprobe p:_fun_6711098_entry /usr/lib/libvitis_ai_library-pointpillars.so:0x9b950
DEBUG:root:### uprobe p:_fun_6989732_entry /usr/lib/libvitis_ai_library-pointpillars.so:0x98fb0
DEBUG:root:### uprobe p:_fun_6989735_entry /usr/lib/libvitis_ai_library-pointpillars.so:0x9bb60
DEBUG:root:### uprobe p:_fun_6989734_entry /usr/lib/libvitis_ai_library-pointpillars.so:0x98d30
DEBUG:root:### uprobe p:_fun_1778374_entry /usr/lib/libvitis_ai_library-openpose.so:0x2bc0
DEBUG:root:### uprobe p:_fun_7506392_entry /usr/lib/libvitis_ai_library-openpose.so:0x3220
DEBUG:root:### uprobe p:_fun_8144855_entry /usr/lib/libvitis_ai_library-clocs.so:0x1a230
DEBUG:root:### uprobe p:_fun_9328147_entry /usr/lib/libvitis_ai_library-centerpoint.so:0x8770
DEBUG:root:### uprobe p:_fun_2531021_entry /usr/lib/libvitis_ai_library-facequality5pt.so:0x2c10
DEBUG:root:### uprobe p:_fun_7046079_entry /usr/lib/libvitis_ai_library-facequality5pt.so:0x34e0
DEBUG:root:### uprobe p:_fun_2906651_entry /usr/lib/libvitis_ai_library-dpu_task.so:0x1a7c0
DEBUG:root:### uprobe p:_fun_2105605_entry /usr/lib/libvitis_ai_library-medicaldetection.so:0x2820
DEBUG:root:### uprobe p:_fun_7223479_entry /usr/lib/libvitis_ai_library-medicaldetection.so:0x2d40
DEBUG:root:### uprobe p:_fun_4198415_entry /usr/lib/libvitis_ai_library-platerecog.so:0x4d60
DEBUG:root:### uprobe p:_fun_6892153_entry /usr/lib/libvitis_ai_library-platerecog.so:0x46b0
DEBUG:root:### uprobe p:_fun_652414_entry /usr/lib/libvitis_ai_library-bcc.so:0x34f0
DEBUG:root:### uprobe p:_fun_1382232_entry /usr/lib/libvitis_ai_library-bcc.so:0x3980
DEBUG:root:### uprobe p:_fun_2600446_entry /usr/lib/libvitis_ai_library-ocr.so:0x5340
DEBUG:root:### uprobe p:_fun_5979864_entry /usr/lib/libvitis_ai_library-ocr.so:0x5800
DEBUG:root:### uprobe p:_fun_59902_entry /usr/lib/libvitis_ai_library-pmg.so:0x3950
DEBUG:root:### uprobe p:_fun_9868632_entry /usr/lib/libvitis_ai_library-pmg.so:0x33f0
DEBUG:root:### uprobe p:_fun_4845822_entry /usr/lib/libvitis_ai_library-ssd.so:0x3150
DEBUG:root:### uprobe p:_fun_227544_entry /usr/lib/libvitis_ai_library-ssd.so:0x3ab0
DEBUG:root:### uprobe p:_fun_227547_entry /usr/lib/libvitis_ai_library-ssd.so:0x34b0
DEBUG:root:### uprobe p:_fun_260502_entry /usr/lib/libvitis_ai_library-3Dsegmentation.so:0x6a30
[... 506 lines cut ...]
            "DPU Arch":"DPUCZDX8G_ISA1_B1600",
            "DPU Frequency (MHz)":325,
            "XRT Frequency (MHz)":100,
            "cu_idx":0,
            "fingerprint":"0x101000056010404",
            "is_vivado_flow":true,
            "name":"DPU Core 0"
        },
        {
            "DPU Arch":"DPUCZDX8G_ISA1_B1600",
            "DPU Frequency (MHz)":325,
            "XRT Frequency (MHz)":100,
            "cu_idx":1,
            "fingerprint":"0x101000056010404",
            "is_vivado_flow":true,
            "name":"DPU Core 1"
        }
    ]
}
root@xilinx-kv260-starterkit-20222:~/app/samples/python#
```

Vivado flow, Petlainux 2022.2

To update the Vitis AI to 3.5 I copied the latest petalinux recipes from this [link](https://github.com/Xilinx/Vitis-AI/tree/master/src/vai_petalinux_recipes) and build the petalinux and booted the image.

Shreyas-NR, 2024-01-11:
Hi @jileilamd,
Were you able to figure out anything from the trace logs?
Please let me know if you want me to try out anything that can be helpful in your debug.
I have attached the python file and the xmodel that I'm using.
Looking forward to hearing from you.

Thanks,
Shreyas
[python.zip](https://github.com/Xilinx/Vitis-AI/files/13902234/python.zip)

Shreyas-NR, 2024-01-22:
Hello @jileilamd 
Do you have any updates? 
It makes sense for me to wait if you respond with at least yes or no.
You are free to tag any other engineers who can resolve this issue. 

I have opened another issue on this repo, you can find the link [here](https://github.com/Xilinx/Vitis-AI/issues/1394)
I read a similar [case ](https://support.xilinx.com/s/question/0D52E00006lLgliSAC/vitis-ai-profiler-with-using-the-dputrd-vivado-flow?language=en_US)on the xiilinx forum that mentions, vaitrace doesn't support vivado flow.
I tried doing Vitis flow for KV260, but I'm stuck with that also, I couldn't get the DPU working using Vitis flow. 

Looking forward to hearing from you.
Best regards,
Shreyas N R