Xilinx/Vitis-AI #1203: Yolov7: [VAIQ_WARN]: Parameter tensor is not quantized: Model::model.105.ia.0.implicit

hoangdung3498, 2023-03-22:
I'm trying to quantize yolov7 by vitisAI 1.4.1. I success when quantize yolov7 which has head is class Detect. But when i quantize with head is class IDetect i have error while calib:
`[VAIQ_WARN]: Parameter tensor is not quantized: Model::model.105.ia.0.implicit

[VAIQ_WARN]: Parameter tensor is not quantized: Model::model.105.im.0.implicit

[VAIQ_WARN]: Parameter tensor is not quantized: Model::model.105.ia.1.implicit

[VAIQ_WARN]: Parameter tensor is not quantized: Model::model.105.im.1.implicit

[VAIQ_WARN]: Parameter tensor is not quantized: Model::model.105.ia.2.implicit

[VAIQ_WARN]: Parameter tensor is not quantized: Model::model.105.im.2.implicit

[VAIQ_WARN]: Quantization is not performed completely, check if model inference function is called!!!`

so i sure because in definition of model have layer follow:part give error in definition of yolov7it seems vitisAI doesn't support nn.Parameter
`##### yolor #####

class ImplicitA(nn.Module):
    def __init__(self, channel, mean=0., std=.02):
        super(ImplicitA, self).__init__()
        self.channel = channel
        self.mean = mean
        self.std = std
        self.implicit = nn.Parameter(torch.zeros(1, channel, 1, 1))
        nn.init.normal_(self.implicit, mean=self.mean, std=self.std)

    def forward(self, x):
        return self.implicit + x
    

class ImplicitM(nn.Module):
    def __init__(self, channel, mean=1., std=.02):
        super(ImplicitM, self).__init__()
        self.channel = channel
        self.mean = mean
        self.std = std
        self.implicit = nn.Parameter(torch.ones(1, channel, 1, 1))
        nn.init.normal_(self.implicit, mean=self.mean, std=self.std)

    def forward(self, x):
        return self.implicit * x
    
##### end of yolor #####`
So have any solution for that ???
thank you very much

tuan97ta, 2023-03-25:
I have the same issue

hoangdung3498, 2023-03-26:
> I have the same issue
seem it is very new problem

Rellek72, 2023-04-20:
Anyone found a solution? It seems to refuse to quantize any custom Parameters,

dungng27, 2023-04-21:
> Anyone found a solution? It seems to refuse to quantize any custom Parameters,

I have to remove these Parameters to make my quantization work

Alexiazzf, 2023-06-13:
@hoangdung3498, VAIQ does not support custom Parameters now. You could change the operator with parameters into nn.Linear or remove these parameters.

Mushegh097, 2023-07-14:
Hi @hoangdung3498 . How do you proceed with the quantization process?
Do you use the pre-trained model of the  yolov7??