Xilinx/Vitis-AI #1206: VITIS-AI-3.0 [VAIQ_ERROR][QUANTIZER_TORCH_CALIB_RESULT_MISMATCH]

jia0511, 2023-03-27:
Node name mismatch is found when loading quantization steps of tensors. Please make sure Vai_q_pytorch version and pytorch version for test mode are the same as those in calibration (or QAT training) mode.

I encountered the above problem in the quantization process, and the problem occurred at 60epoch. How should I solve this problem?
[qat_error.txt](https://github.com/Xilinx/Vitis-AI/files/11077474/qat_error.txt)

gaoxinghua951, 2023-06-08:
I have the same problem. Have you solved it?

Alexiazzf, 2023-06-14:
Hi, @jia0511 

Issue https://github.com/Xilinx/Vitis-AI/issues/1216 runs into the same error. It needs to specify the same target when you do quantization calibration and test.
So the script to run should be, for example:

python vitis_min.py --target DPUCZDX8G_ISA1_B1024
python vitis_min.py --quant_mode test --export --target DPUCZDX8G_ISA1_B1024