Xilinx/Vitis-AI #1208: Register Aten Custom Ops

dungng27, 2023-03-28:
Hi, I'm trying to quantize and compile a PyTorch model with some Aten operations not supported by Vitis-AI yet. Specifically, I'm deploying the model but some errors occured:
```
[VAIQ_NOTE]: Quant config file is empty, use default quant configuration

[VAIQ_NOTE]: Quantization test process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cpu'.

[VAIQ_NOTE]: =>Parsing RotatedSegmentDetector...

[VAIQ_NOTE]: Start to trace model...
/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/torch/nn/functional.py:2359: UserWarning: __floordiv__ is deprecated, and its behavior will change in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor').
  _verify_batch_size([input.size(0) * input.size(1) // num_groups, num_groups] + list(input.size()[2:]))

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 403/403 [00:00<00:00, 2273.19it/s, OpInfo: name = return_0, type = Return]                           

[VAIQ_WARN]: The quantizer recognize new op `aten::mul_` as a float operator by default.

[VAIQ_WARN]: The quantizer recognize new op `aten::group_norm` as a float operator by default.

[VAIQ_WARN]: The quantizer recognize new op `aten::clamp_` as a float operator by default.

[VAIQ_WARN]: The quantizer recognize new op `clamp` as a float operator by default.

[VAIQ_NOTE]: =>Doing weights equalization...
/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/optimization/commander.py:454: RuntimeWarning: divide by zero encountered in true_divide
  scale = np.where(sqrt_of_ranges != 0, range_0 / sqrt_of_ranges, scale)
/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/optimization/commander.py:454: RuntimeWarning: invalid value encountered in true_divide
  scale = np.where(sqrt_of_ranges != 0, range_0 / sqrt_of_ranges, scale)

[VAIQ_NOTE]: =>Quantizable module is generated.(/workspaces/PheNet_Vitis-AI/quant_model/RotatedSegmentDetector.py)

[VAIQ_NOTE]: =>Get module with quantization.

[VAIQ_NOTE]: =>Converting to xmodel ...
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0328 02:17:35.101768 21213 tool_function.cpp:171] [UNILOG][WARNING] The operator named RotatedSegmentDetector__RotatedSegmentDetector_MobileNetV3_backbone__InvertedResidual_layer1__SELayer_se__ConvModule_conv2__HSigmoid_activate__20467, type: aten::clamp_, is not defined in XIR. XIR creates the definition of this operator automatically. You should specify the shape and the data_type of the output tensor of this operation by set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)
W0328 02:17:35.176827 21213 tool_function.cpp:171] [UNILOG][WARNING] The operator named RotatedSegmentDetector__RotatedSegmentDetector_RotatedFCOSHead_bbox_head__ConvModule_cls_convs__ModuleList_0__GroupNorm_gn__input_309, type: aten::group_norm, is not defined in XIR. XIR creates the definition of this operator automatically. You should specify the shape and the data_type of the output tensor of this operation by set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)
W0328 02:17:35.193320 21213 tool_function.cpp:171] [UNILOG][WARNING] The operator named RotatedSegmentDetector__RotatedSegmentDetector_RotatedFCOSHead_bbox_head__22828, type: cast, is not defined in XIR. XIR creates the definition of this operator automatically. You should specify the shape and the data_type of the output tensor of this operation by set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)
W0328 02:17:35.193936 21213 tool_function.cpp:171] [UNILOG][WARNING] The operator named RotatedSegmentDetector__RotatedSegmentDetector_RotatedFCOSHead_bbox_head__22831, type: clamp, is not defined in XIR. XIR creates the definition of this operator automatically. You should specify the shape and the data_type of the output tensor of this operation by set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)
F0328 02:17:35.193950 21213 wrapper.cpp:111] [UNILOG][FATAL][PYXIR_INVALID_DATA_TYPE][] Unsupported data type!
*** Check failure stack trace: ***
Aborted (core dumped)
```
Here is the script i run to quantize and deploy the model:
```
from mmcv import collect_env
collect_env()

# Check MMRotate installation
import mmrotate
print(mmrotate.__version__)

# Check MMDetection installation
import mmdet
print(mmdet.__version__)

# Check mmcv installation
from mmcv.ops import get_compiling_cuda_version, get_compiler_version
print(get_compiling_cuda_version())
print(get_compiler_version())

# from utils.postprocess import *
import glob
import os.path as osp
import shutil
import os
[... 109 lines cut ...]
    # for iteraction, images in tqdm(
    #   enumerate(val_loader), total=len(val_loader)):
    #     output = quantized_model(images)

    output = quantized_model(dummy_input)

    # for path in tqdm(images):
    #     input_tensor = file2tensor(path)
    #     output = quantized_model(input_tensor)

    if mode == 'calib':
        quantizer.export_quant_config()
    else:
        quantizer.export_xmodel(deploy_check=False, output_dir=output_dir)
```
with the command: 
```
python test_code/inference_pth.py -q test \
-cf /workspaces/PheNet_Vitis-AI/models/multihead/rotated_fcos_kld_mbnv3_fpn_1x_dota_le90.py \
-c /workspaces/PheNet_Vitis-AI/models/multihead/multihead.pth \
-o /workspaces/PheNet_Vitis-AI/quant_model
```
I'm using Vitis-AI 2.5 for stability. 
I read the docs about Register Custom Operation but I don't know how to apply this workflow to register these custom Aten ops. Could someone show me how?
Many thanks.

dungng27, 2023-03-30:
It turns out that I can modify these ops by using torch functions rather than tensor's functions and register these. The problem is solved!

dungng27, 2023-04-06:
> It turns out that I can modify these ops by using torch functions rather than tensor's functions and register these. The problem is solved!
This is just a temporary fix and not applicable to other ops. I wonder if there is a better work around.

manudwd, 2023-09-29:
@dungng27  
> It turns out that I can modify these ops by using torch functions rather than tensor's functions and register these. The problem is solved!

Can you elaborate a little more on this with a dummy example? I'd really appreciate it.

acg93-pixel, 2023-11-15:
Did you manage to add validation for quantized model from MMEngine? Can you share please?

SM2XWe1L, 2024-08-26:
> I'm also stumped on this, with the specific operators:
> 
> ```
> aten::meshgrid
> aten::split_with_sizes
> ```

Hello，i also encounter the same problem as yours,i want to know if you have solved it?
Thirsty for your reply!

acg93-pixel, 2024-08-26:
Hi, you have to replace these operations in your model definition with supported ones.

SM2XWe1L, 2024-08-26:
> Hi, you have to replace these operations in your model definition with supported ones.

Thanks for your kind reply! 
But i still have a problem that i have found meshgrid in make_anchors function,but i can't catch where the split_with_sizes is , could you tell me where aten::split_with_sizes is used?

Thanks.

acg93-pixel, 2024-08-26:
Well, not sure which model and repository version are you using. Perhaps in PyCharm try "Find in Files" option with just "split_with_sizes".
In my case, I used RTMDet-Ins in MMdetection, but "split_with_sizes" occurred in post-processing, which quantizer anyhow  doesn't put in the model. So, everything after the model forward function, you'd have to implement yourself.

SM2XWe1L, 2024-08-26:
> Well, not sure which model and repository version are you using. Perhaps in PyCharm try "Find in Files" option with just "split_with_sizes". In my case, I used RTMDet-Ins in MMdetection, but [...]

fully thanks!
Actually i am using vitis-ai to quantize yolov8 model , i will try to find the op in post-processing.