Xilinx/Vitis-AI #1210: MaskRCNN deployment not currently supported?

jia0511, 2023-03-30:
I see that you provide maskRCNN model in modezoo, but you haven't seen the relevant deployment code. Is it not supported at present？

daniperfer, 2025-04-03:
Hello, 
I have a similar question. In version 3.0 of the model zoo, there is the model pt_MaskRCNN_coco_800_800_240G_3.0. The model zoo provides float and quantized weights to run the model on GPU, but it also includes the quantization script.

I have included a line in the script to try to obtain the exported xmodel file, thinking of deploying the model on the FPGA later, but when I try to export the model, errors are generated in the logs. Is the model pt_MaskRCNN_coco_800_800_240G_3.0 prepared for deployment on FPGA, or would something need to be done to achieve it?

quentonh, 2025-04-03:
I am fairly certain that we never successfully deployed MaskRCNN on the DPU.  If I recall correctly, the inclusion of that model in the release was an oversight that should have been fixed.  There were a handful of models that could be quantized, but not deployed.

daniperfer, 2025-04-03:
Thank you for your quick response, @quentonh.
Our intention was to deploy a semantic segmentation model on the DPU. But if the MaskRCNN model is not appropriate for deployment on DPU, then we will have to look for other options for segmentation with DPU.
Do you know if there is any segmentation model that has been successfully deployed on DPU?

quentonh, 2025-04-03:
I just remembered now (I have forgotten as much as I ever knew) that you can simply check the Model Zoo spreadsheet.  If the model wasn't deployed, there won't be a benchmark for it.  There, you will find segmentation models that have been deployed.

https://xilinx.github.io/Vitis-AI/3.0/html/docs/workflow-model-zoo.html

quentonh, 2025-04-03:
I think that we left some of the non-deployable models in the repo because they were also used for an inference solution for our CPUs, or something like that.  I know, it is weird and a bit awkward, so please accept our apologies for that.  The Model Zoo spreadsheet should be your guide for what can be deployed.

daniperfer, 2025-04-03:
Thank you, @quentonh. We will take a look at what other models in the spreadsheet have benchmark reports. Specifically on ZCU102, which is the hardware we have.