Xilinx/Vitis-AI #1212: Error response from daemon: No such image: xilinx/vitis-ai-gpu:latest

Truthonlyone, 2023-04-01:
Vitis AI version: 1.4
System: Ubuntu 16.04
GPU: NVIDIA GeForce RTX 2080 SUPER

I want to speed the quantization process, so I tried to install the Vitis AI 1.4 on my computer with Ubuntu 16.04, while there are some isuues raised. 

At the first time I run "sudo ./docker_build_gpu.sh", my computer is freeze in the halfway，the system time is freese, and the mouse and there is no response from the mouse and keyboard. So I forcibly shut down the computer.

Then, I restart the computer and execute the command "sudo ./docker_build_gpu.sh", the error
[image]
 "Error response from daemon: No such image: xilinx/vitis-ai-gpu:latest" occured.

The log is in the attachment.
[log.txt](https://github.com/Xilinx/Vitis-AI/files/11129884/log.txt)

How can I fix it? Please help me, thx a lot!

Truthonlyone, 2023-04-02:
hey, 
  I ran the command "sudo ./docker_build_gpu.sh" again, and the issue above has not happened. 
  Maybe it's about network.
  But there is a new problem, please see the pitcture. 
  
[image]

  What should I do?

janifer112x, 2023-04-04:
Hi @Truthonlyone，
 I will have a try and update you later

Truthonlyone, 2023-04-05:
> Looking forward to your reply!