Xilinx/Vitis-AI #1215: DPU kernet driver not working for Vitis-AI 3.0 in petalinux 2022.2 project.

AlbertaBeef, 2023-04-04:
DPU kernet driver not working for Vitis-AI 3.0 in petalinux 2022.2 project.

Adding the three CONFIG_vitis-ai-library* packages to project per instructions here:
https://github.com/Xilinx/Vitis-AI/blob/v3.0/src/vai_petalinux_recipes/README.md

Added following content to my project's project-spec/meta-user directory:
https://github.com/Xilinx/Vitis-AI/tree/v3.0/src/vai_petalinux_recipes/recipes-vitis-ai

Build completes fine, but execution generates errors:

root@zub1cg-sbc-2022-2:~# xdputil query
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0404 13:28:17.234927   692 xrt_device_handle.cpp:44] Check failed: !get_factory_methods().empty()
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:   692 Aborted                 /usr/bin/python3 -m xdputil $*

Not certain what is causing this ... are there other steps to do to configure petalinux project (beyond what is described on github) ?

Complete log (with dmesg output):

root@zub1cg-sbc-2022-2:~# dmesg --clear
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~# xmutil loadapp avnet-zub1cg-benchmark
[  249.792948] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /fpga-full/firmware-name
[  249.803072] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /fpga-full/resets
[  249.813468] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/afi0
[  249.822977] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_gpio_0
[  249.832990] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_gpio_1
[  249.842998] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_gpio_2
[  249.853004] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_iic_0
[  249.862926] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_iic_1
[  249.872847] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_iic_2
[  249.882768] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_intc_0
[  249.892776] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_quad_spi_0
[  249.903131] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_uartlite_0
[  249.913487] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/system_management_wiz_0
[  250.569133] xadc a0090000.system_management_wiz: IRQ index 0 not found
[  250.614628] zocl-drm axi:zyxclmm_drm: IRQ index 32 not found
avnet-zub1cg-benchmark: loaded to slot 0
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~# dmesg
[  249.792948] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /fpga-full/firmware-name
[  249.803072] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /fpga-full/resets
[  249.813468] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/afi0
[  249.822977] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_gpio_0
[  249.832990] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_gpio_1
[  249.842998] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_gpio_2
[  249.853004] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_iic_0
[  249.862926] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_iic_1
[  249.872847] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_iic_2
[  249.882768] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_intc_0
[  249.892776] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_quad_spi_0
[  249.903131] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/axi_uartlite_0
[  249.913487] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/system_management_wiz_0
[  250.547903] irq-xilinx: mismatch in kind-of-intr param
[  250.547924] irq-xilinx: /axi/interrupt-controller@a0060000: num_irq=32, sw_irq=0, edge=0x1
[  250.566890] a0080000.serial: ttyUL1 at MMIO 0xa0080000 (irq = 75, base_baud = 0) is a uartlite
[  250.569133] xadc a0090000.system_management_wiz: IRQ index 0 not found
[  250.614289] [drm] Probing for xlnx,zocl
[  250.614628] zocl-drm axi:zyxclmm_drm: IRQ index 32 not found
[  250.621117] [drm] FPGA programming device pcap founded.
[  250.624410] [drm] PR[0] Isolation addr 0x0
[  250.624944] [drm] Initialized zocl 0.0.0 00000 for axi:zyxclmm_drm on minor 0
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~# dmesg --clear
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~# cat /etc/vart.conf
firmware: /lib/firmware/xilinx/avnet-zub1cg-benchmark/avnet-zub1cg-benchmark.xclbin
root@zub1cg-sbc-2022-2:~# ls -la /lib/firmware/xilinx/avnet-zub1cg-benchmark/
total 2140
drwxr-xr-x 2 root root    4096 Mar  9  2018 .
drwxr-xr-x 5 root root    4096 Mar  9  2018 ..
-rw-r--r-- 1 root root 2980860 Mar  9  2018 avnet-zub1cg-benchmark.bit.bin
-rw-r--r-- 1 root root   20402 Mar  9  2018 avnet-zub1cg-benchmark.dtbo
-rw-r--r-- 1 root root   41350 Mar  9  2018 avnet-zub1cg-benchmark.xclbin
-rw-r--r-- 1 root root      43 Mar  9  2018 shell.json
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~# xdputil query
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0404 13:28:17.234927   692 xrt_device_handle.cpp:44] Check failed: !get_factory_methods().empty()
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:   692 Aborted                 /usr/bin/python3 -m xdputil $*
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~#
root@zub1cg-sbc-2022-2:~# dmesg
[  284.032136] audit: type=1701 audit(1680614897.244:3): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=692 comm="python3" exe="/usr/bin/python3.9" sig=6 res=1
root@zub1cg-sbc-2022-2:~#

theo06v, 2023-04-07:
Hi,
I have the same problem as you, did you find a solution?

AlbertaBeef, 2023-04-09:
No solution, but I found a workaround, documented here : [https://avnet.me/avnet-zub1cg-2022.2-part3](https://avnet.me/avnet-zub1cg-2022.2-part3) 

```
root@zub1cg-sbc-2022-2:~# mkdir rpm
root@zub1cg-sbc-2022-2:~# cd rpm
root@zub1cg-sbc-2022-2:~/rpm# wget https://github.com/Xilinx/Vitis-AI/raw/v3.0/board_setup/mpsoc/2022.2/libunilog-3.0.0-r86.aarch64.rpm
root@zub1cg-sbc-2022-2:~/rpm# wget https://github.com/Xilinx/Vitis-AI/raw/v3.0/board_setup/mpsoc/2022.2/libxir-3.0.0-r86.aarch64.rpm
root@zub1cg-sbc-2022-2:~/rpm# wget https://github.com/Xilinx/Vitis-AI/raw/v3.0/board_setup/mpsoc/2022.2/libtarget-factory-3.0.0-r86.aarch64.rpm
root@zub1cg-sbc-2022-2:~/rpm# wget https://github.com/Xilinx/Vitis-AI/raw/v3.0/board_setup/mpsoc/2022.2/libvart-3.0.0-r86.aarch64.rpm
root@zub1cg-sbc-2022-2:~/rpm# wget https://github.com/Xilinx/Vitis-AI/raw/v3.0/board_setup/mpsoc/2022.2/libvitis_ai_library-3.0.0-r86.aarch64.rpm

root@zub1cg-sbc-2022-2:~/rpm# rpm -ivh --force libunilog-3.0.*-r*.aarch64.rpm
root@zub1cg-sbc-2022-2:~/rpm# rpm -ivh --force libxir-3.0.*-r*.aarch64.rpm
root@zub1cg-sbc-2022-2:~/rpm# rpm -ivh --force libtarget-factory-3.0.*-r*.aarch64.rpm
root@zub1cg-sbc-2022-2:~/rpm# rpm -ivh --force libvart-3.0.*-r*.aarch64.rpm
root@zub1cg-sbc-2022-2:~/rpm# rpm -ivh --force libvitis_ai_library-3.0.*-r*.aarch64.rpm

root@zub1cg-sbc-2022-2:~/rpm# cd ..
root@zub1cg-sbc-2022-2:~#
```

quentonh, 2023-04-11:
@theo06v If you are building with Vitis, please be sure to remove the vart_3.0_vivado.bb file from the Petalinux recipes folder.  We will update the instructions to provide additional clarity.  @AlbertaBeef tested this and it fixed the problem for his build.

closed 2023-04-11 by AlbertaBeef

theo06v, 2023-04-13:
I'm using Vivado flow. There seems to be a problem with the DPU during the boot. 
root@xilinx-zcu104-trd:~# dmesg | grep dpu
[    4.698888] xlnx-dpu 80000000.dpuczdx8g: IRQ index 0 not found
[    4.704718] xlnx-dpu: probe of 80000000.dpuczdx8g failed with error -22

Should I update the device tree in "system-user.dtsi"?
Also, should I add "CONFIG_dpu" in "user-rootfsconfig"?

theo06v, 2023-04-17:
Thank you for your reply. However, I already use the file "vart_3.0_vivado.bb".
I also use the "helper_build_bsp.sh" script to create my petalinux platform (I modified the .bsp file to match the ZCU104).

quentonh, 2023-04-18:
@theo06v In the Vivado design context, you should not be seeing messages related to ZOCL nor to DeviceTree Overlays unless the Runtime is built for the Vitis flow, and yet you state that you are seeing the same messages as AlbertaBeef.  For Vivado designs, XRT is not used.  Is there a possibility that you have the XRT source tree pulled into your Petalinux project?  What is your device target for this design, Zynq Ultrascale+?

theo06v, 2023-04-19:
I had the same error when running "xdputil query" command.
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0404 13:28:17.234927 692 xrt_device_handle.cpp:44] Check failed: !get_factory_methods().empty()

My device target for this design is Zynq Ultrascale+ and i have disabled zocl and xrt packages.

quentonh, 2023-04-20:
@theo06v Have you been able to investigate the suggested solutions provided here?

https://support.xilinx.com/s/question/0D54U00005kNQeNSAW/dpucontrollercpp44-check-failed-thefactorymethodsempty?language=en_US

Yuxin-Yu, 2024-04-16:
Hello @AlbertaBeef , I cannot find the xmutil command in my PetaLinux, and the loadapp option does not exist in the xdputil command.Do you know the reasons ?