Xilinx/Vitis-AI #1217: MatMuls without Lambda in TensorFlow 2

Rellek72, 2023-04-07:
`matmul` is listed as a [supported operation for TensorFlow 2](https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Operators-Supported-by-TensorFlow) and mapped to a conv2d in the DPU. However, matmul in TensorFlow is implemented using a Lambda layer, which is not supported for quantization and thus compiled for the CPU.

Considering the following toy model with input shape [32, 62, 64]:

```
def my_model():
    dim1 = 62
    dim2 = 64
    classes = 10

    inputs = layers.Input(shape=(dim1, dim2))
    
    product = tf.linalg.matmul(inputs, inputs, transpose_b=True)
    pooled = layers.GlobalAveragePooling1D()(product)
    outputs = layers.Dense(classes, activation="softmax")(pooled)

    model = tf.keras.Model(inputs=inputs, outputs=outputs)

    return model
```

This is the model summary after training (As you can see, the matmul is listed as a TFOpLambda):
```
__________________________________________________________________________________________________
Model: "model"
__________________________________________________________________________________________________
 Layer (type)                   Output Shape         Param #     Connected to                     
==================================================================================================
 input_1 (InputLayer)           [(None, 62, 64)]     0           []                               
                                                                                                  
 tf.linalg.matmul (TFOpLambda)  (None, 62, 62)       0           ['input_1[0][0]',                
                                                                  'input_1[0][0]']                
                                                                                                  
 global_average_pooling1d (Glob  (None, 62)          0           ['tf.linalg.matmul[0][0]']       
 alAveragePooling1D)                                                                              
                                                                                                  
 dense (Dense)                  (None, 10)           630         ['global_average_pooling1d[0][0]'
                                                                 ]                                
                                                                                                  
==================================================================================================
Total params: 630
Trainable params: 630
Non-trainable params: 0
__________________________________________________________________________________________________
```

And the following error is given when quantizing:
```
[VAI INFO] Layer tf.linalg.matmul(<class 'keras.layers.core.tf_op_layer.TFOpLambda'>) is not supported by DPU,
it will not be quantized and may be mapped to run on CPU or other IPs. Please see User Guide for list of supported 
operations and APIs of vai_q_tensorflow2.
```

Using the `tf.matmul` operator gives the same behavior. How can I use a matrix multiply in my model in a way that can be serialized and compiled correctly for the DPU?

staileng, 2023-04-19:
Were you able to find a solution to this problem? Scratching my head right now with what looks like the same issue

Rellek72, 2023-05-26:
Bumping to hope for some assistance, this is a pretty fundamental operation for custom neural networks in TensorFlow that doesn't seem to work at all.

Rellek72, 2023-07-03:
Closing this issue, as it has been fixed in Vitis AI 3.5. However, `tf.linalg.mamtul` still remains unusable as it's unable to compile. Continuing in [this issue](https://github.com/Xilinx/Vitis-AI/issues/1276).

closed 2023-07-03 by Rellek72