Xilinx/Vitis-AI #1219: Cannot register custom op

dungng27, 2023-04-12:
Hi I'm following the guideline here [https://github.com/Xilinx/Vitis-AI/tree/2.0/demo/Custom_OP_Demo/op_add](url) to register my clamp custom op because **Vitis-AI 1.4.1.978** does not support this op.
However, when I run these code:
```
$ ipython
import xir
g = xir.Graph("simple_graph")
a = g.create_op("a", "data", {"shape": [1,2,2,4], "data_type": "FLOAT32"});
add = g.create_op("clamp_op", "clamp",  {"shape": [1,2,2,4], "data_type": "FLOAT32"}, {"input": [a]})
root = g.get_root_subgraph()
root.create_child_subgraph()
user_subgraph = root.merge_children(set([g.get_leaf_subgraph(a), g.get_leaf_subgraph(b)]))
cpu_subgraph = root.merge_children(set([g.get_leaf_subgraph(add)]))
user_subgraph.set_attr("device", "USER")
cpu_subgraph.set_attr("device", "CPU")
g.serialize("/tmp/clamp.xmodel")
```
it shows the error:
```
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0412 01:35:23.702582 29711 op_def_factory_imp.cpp:101] [UNILOG][FATAL][XIR_UNREGISTERED_OP][Unregistered operator!] clamp
*** Check failure stack trace: ***
Aborted (core dumped)
```
Here is my clamp op implementation file:
```
// clamp.cpp
#include <cmath>
#include <iostream>
#include <vart/op_imp.h>

struct MyClampOp : public vart::experimental::OpImpBase {
    public:
        MyClampOp(xir::Op* op, xir::Attrs* attrs) 
        : vart::experimental::OpImpBase{op, attrs} {
            min_ = op->get_attr<float>("min");
            max_ = op->get_attr<float>("max");
        }

        int calculate(vart::experimental::simple_tensor_buffer_t<float> output,
                    std::vector<vart::experimental::simple_tensor_buffer_t<float>> inputs) {
            auto inputs_num_of_elements = inputs[0].tensor->get_element_num();
            auto output_num_of_elements = output.tensor->get_element_num();

            CHECK_EQ(inputs_num_of_elements, output_num_of_elements);
            CHECK_EQ(inputs.size(), 1);

            for (auto i = 0u; i < output.mem_size / sizeof(float); ++i) {
                output.data[i] = std::max(std::min(inputs[0].data[i], min_), max_);
            }
            return 0;
        }

    private:
        float min_;
        float max_;
};

extern "C" vart_op_imp_t vart_init_op_imp(const xir_op_t op) {
    return vart::experimental::make_vart_opt_imp<MyClampOp>();
}

int DEF_XIR_OP_IMP(MyClampOp);

```
which is in the same folder with my Makefile:
```
OUTPUT_DIR := /tmp/lib

all: $(OUTPUT_DIR) $(OUTPUT_DIR)/libvart_op_imp_clamp.so

$(OUTPUT_DIR):
	mkdir -p $@

$(OUTPUT_DIR)/clamp.o: clamp.cpp
	$(CXX) -std=c++17 -fPIC -c -o $@ -I. -I=/usr/include -Wall  -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=0 $<

$(OUTPUT_DIR)/libvart_op_imp_clamp.so:  $(OUTPUT_DIR)/clamp.o
	$(CXX) -Wl,--no-undefined -shared -o $@ $+ -L=/usr/lib  -lglog -lvitis_ai_library-runner_helper -lvart-runner -lxir

.PHONY: clean

clean:
	rm -f $(OUTPUT_DIR)/*.so $(OUTPUT_DIR)/*.o
```
After I run `make all` then `ipython` it results in the error above.
My Vitis-AI version is **1.4.1.978**
Do I register it incorrectly or do I miss something?
I've been struggling with this for long. I really appreciate any support.

lishixlnx, 2023-04-19:
The url you gave is old and can't be opened.
From the error message, it seems the lib can't be found.
Please copy the compiled libxxx.so to the destination board path : /usr/lib  
and try again.

dungng27, 2023-04-20:
> The url you gave is old and can't be opened. From the error message, it seems the lib can't be found. Please copy the compiled libxxx.so to the destination board path : /usr/lib and try again.

Thanks for replying. It turns out that I have to create .so file in the board not in my local computer.

closed 2023-04-20 by dungng27