Xilinx/Vitis-AI #1222: undefined symbol: g_unix_get_passwd_entry : Error when quantized tf_yolov4_coco_416_416_60.3G_2.5

Satoski-H, 2023-04-20:
Hi all,
I'm trying to quantize tf_yolov4_coco_416_416_60.3G_2.5 with Vitis-AI 2.5 on vitis-ai-cpu container.
I did it following to README.md.
But I got errors as below.

"""""""""""""""""""""""""""""""""
(vitis-ai-tensorflow) Vitis-AI /workspace/tf_yolov4_coco_416_416_60.3G_2.5/code/quantize > bash ./quantize_416.sh
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 260, in _parse_input_fn
    module = __import__(input_fn_str.rsplit('.', 1)[0], fromlist=True)
  File "./input_fn_416.py", line 6, in <module>
    import cv2
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/cv2/__init__.py", line 153, in bootstrap
    native_module = importlib.import_module("cv2")
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/../lib/libgio-2.0.so.0: undefined symbol: g_unix_get_passwd_entry

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/vai_q_tensorflow", line 11, in <module>
    sys.exit(run_main())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 1255, in run_main
    app.run(main=my_main, argv=[sys.argv[0]] + unparsed)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/platform/app.py", line 40, in run
    _run(main=main, argv=argv, flags_parser=_parse_flags_tolerate_undef)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/app.py", line 312, in run
    _run_main(main, args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/app.py", line 258, in _run_main
    sys.exit(main(argv))
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 1254, in <lambda>
    my_main = lambda unused_args: main(unused_args, FLAGS)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 821, in main
    input_fn = _parse_input_fn(flags.input_fn)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 263, in _parse_input_fn
    raise ValueError('Fail to import input_fn, error: ', e)
ValueError: ('Fail to import input_fn, error: ', ImportError('/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/../lib/libgio-2.0.so.0: undefined symbol: g_unix_get_passwd_entry',))
"""""""""""""""""""""""""""""""""

I installed glib (pip install glib) but it didn't make any effect on it.
How can I solve it?
Any help would be appreciated.

Thank you,

zhenzhen-AMD, 2023-07-06:
Hi @Satoski-H ,
1. What is the version of glib?
2. I have not reproduced the error. The steps to reproduce the issue are as follows:
- conda activate vitis-ai-tensorflow
- pip install opencv-python==4.2.0.34 
- pip install imgaug
- bash quantize_416.sh

I hope this helps! Let me know if you have any further questions.

Satoski-H, 2023-07-07:
Hi @zhenzhen-AMD

Thank you for your great help!

The way you showed me worked fine.

The version of glib I installed was 1.0.0.
But I didn't export it into my environment so this time there was no glib installed.

Then it seemed that  my probrem was caused by the version of opencv-python, right?

Kind regards.

zhenzhen-AMD, 2023-07-07:
Hi @Satoski-H ,

I can't determine the specific reason for the error.
However, according to the error message you provided, it seems the error may be due to a missing glib. 
In addition, this project also depends on the opencv-python and imgaug. 

Best Regard.

Satoski-H, 2023-07-07:
Hi @zhenzhen-AMD

Thank you for your reply.

My error message appeared to say there was no glib though, it was okay without installing it.
Anyway thank you again for all of your kindness.

Kind regards.

closed 2023-07-07 by Satoski-H

zhenzhen-AMD, 2023-07-07:
You are welcome.