Xilinx/Vitis-AI #1224: No xclbin.ex.run_summary is produced by vaitrace

Edocit, 2023-04-21:
Hi, I'm using the simple DPU design in Figure "design" with the presets in Figure "zynqultrascale_preset".
[image]
[image]

 I followed what written in the vaitrace documentation. I've used Vivado 2022.2 to build the XSA platform in figure and petalinux 2022.2 to build the linux project and Vitis-AI 3 (github master branch) for the acceleration framework. The project was created starting from the BSP of the ZCU102 for the version 2022.2 (just to be clear petalinux-create -t project -s zcu102_2022.2.bsp -n project_name). **For the inference part it works like charm** but when I want to trace the 2 core execution 2 thing happened:

1) I have a float division by zero error in "efficiency" variable computation as reported in figure below
[image]

Clearly, the error was caused by this line, and suppose just for a moment we can hardcode efficiency to 0 just to bypass the previous issue
[image]

2) With this modification no other errors are thrown and the vaitrace for the resnet50_mt_py with 2 threads finish its execution as reported 
[image]

The problem here is that only the following files were created and no xclbin.run_summary was created.
[image]

Can you please help me to get trace files for multicore DPU. **It is very important to solve this issue as soon as possible** 

Thank you,
Edoardo

jileilxinlix, 2023-05-17:
hi @Edocit 
q1: maybe you not open CONFIG_FTRACE ,if your log show:
WARNING:root:No such file or directory: '/sys/kernel/debug/tracing/trace_clock'
WARNING:root:CPU function tracing feature is disabled
so vaitrace can not get clock.

q2: xclbin.run_summary has been replaced by xrt.run_summary, but i can not find your result has the file. 
we follow  your flow and execute some command and xrt.run_summary has beeb create.

qianglin-xlnx, 2023-06-25:
Closing since no activity for more than 1 month, please reopen if you still have question, thanks

closed 2023-06-25 by qianglin-xlnx

Shreyas-NR, 2024-01-10:
Hi @Edocit ,
I'm also having the same issue, vaitrace not generating the .run_summary file. 
Can you help me by sharing what did you do to get it working?