Xilinx/Vitis-AI #1239: CANNOT OPEN BUILT DOCKER??????!!!!

HankIC, 2023-05-16:
I build docker successfully, but I cannot open it???? I build docker use sudo ./docker_build.sh -t gpu -f pytorch,
[image]
[image]

1411030449, 2023-07-30:
Hi, may I ask you to solve it? I also encountered the same problem, I built the image but could not open it:
qq@fqq-MS-7D17:~/Vitis-AI$ docker images
REPOSITORY                        TAG                                 IMAGE ID       CREATED        SIZE
xilinx/vitis-ai-pytorch-gpu       3.0.0.001                           b39be7c1f120   3 hours ago    16.7GB
xilinx/vitis-ai-pytorch-gpu       latest                              b39be7c1f120   3 hours ago    16.7GB
xilinx/vitis-ai-opt-pytorch-gpu   3.0.0.001                           53b716ebe58b   3 hours ago    13.4GB
xilinx/vitis-ai-opt-pytorch-gpu   latest                              53b716ebe58b   3 hours ago    13.4GB
xiinx/vitis-ai-gpu-base           latest                              5979eeb9481e   3 hours ago    5.97GB
nvidia/cuda                       11.3.1-cudnn8-runtime-ubuntu20.04   41ab429ac5e7   5 weeks ago    3.31GB
hello-world                       latest                              9c7a54a9a43c   2 months ago   13.3kB
fqq@fqq-MS-7D17:~/Vitis-AI$ sudo ./docker_run.sh xilinx/vitis-ai-pytorch-gpu:latest
Unable to find image 'xilinx/vitis-ai-pytorch-gpu:latest' locally
docker: Error response from daemon: pull access denied for xilinx/vitis-ai-pytorch-gpu, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
fqq@fqq-MS-7D17:~/Vitis-AI$

janifer112x, 2023-07-31:
Hi @HankIC @1411030449 , Thanks for bringing out this issue, let me try to reproduce it and update you later.

1411030449, 2023-08-01:
@janifer112x 

Hi, I used Ubuntu:20.04.05, Nvidia-driver: 470.199.02, cuda:11.4, GPU: NVIDIA GEFORCE RTX3060, NVIDIA-container-toolkit:1.13.5-1, docker:desktop-24.0.5 to build Vitis AI-3.0, but I have failed many times. I operate according to the installation instructions:https://xilinx.github.io/Vitis-AI/3.0/html/docs/install/install.html. Thank you very much for your help.

janifer112x, 2023-08-01:
Hi @1411030449 , @HankIC 
Sorry for the late reply.  I saw you are trying to use sudo to launch VitisAI gpu docker, could you please try with ./docker_run.sh  xilinx/vitis-ai-pytorch-gpu:3.0.0.001 instead?   during the docker_build.sh, it did not create latest tag any more. please let me know if this works.

1411030449, 2023-08-01:
@janifer112x 
Thank you very much for your help, I rebuild it again. I used the command: ./docker_run.sh xilinx/vitis-ai-pytorch-gpu:3.0.0.001 to start it successfully, but I meet the error: **_bash: /usr/local/bin/gosu: No such file or directory_**
I don't know how to solve it. Thanks again.

janifer112x, 2023-08-01:
Glad to see that you can launch the docker, @1411030449 
the gosu was installed during the base image build phase, I cannot reproduce it from my side, could you please check the build log see if there is anything goes wrong with the gosu installation during the build base image? or maybe you can try to run the command to see if it works: curl -sSkLo <some othr path>/gosu "https://gitclone.com/github.com/tianon/gosu/releases/download/$GOSU_VERSION/gosu-$(dpkg --print-architecture)"**

docker/common/install_base.sh:    && **curl -sSkLo /usr/local/bin/gosu "https://gitclone.com/github.com/tianon/gosu/releases/download/$GOSU_VERSION/gosu-$(dpkg --print-architecture)"** \
docker/common/install_base.sh:    && chmod +x /usr/local/bin/gosu
docker/conda/bashrc:  exec /usr/local/bin/gosu vitis-ai-user "$@"

1411030449, 2023-08-01:
@janifer112x 
Thanks very much.
build log shows: **_fatal : not a git repository ( or any parent directory up to the mount point / ) 
Stop at the file system boundary ( GIT _ DISCOVERY _ ACROSS _ FILESYSTEM is not set )._**
but it did not stop building image. 
In addition, The command you mentioned above does not solve the problem.
What can I do next?
Thanks very much.

1411030449, 2023-08-01:
@janifer112x 
I rebuilt it again, and this time seems successful. It can successfully run Vitis-AI, but there will also be a wrong: **_Error response from daemon: pull access denied for xilinx/vitis-ai-pytorch-gpu, repository does not exist or may require 'docker login': denied:_** **_requested access to the resource is denied_**, but it does not affect the normal use.
Thanks very much! You are Superman.

janifer112x, 2023-08-01:
@1411030449 ，glad to hear that. you can ignore the docker pull error for now, as it tries to pull latest images from dockerhub, but we don't have  nvidia gpu docker published in  docker hub. I will see how to get rid of the faking error  later, if no objection, I will close this issue , feel free to submit a new issue in the future